sfstar/hass-victron #1: Error: Configuration cannot be loaded: {"message":"Invalid handler specified"}

shaman171, 2022-11-29:
Hello, 

at first thank you a lot for doing this one! :) 

Never the less I have an issue. I installed via HASS and when I want to add the integration it gives me the error: 
_Configuration cannot be loaded: {"message":"Invalid handler specified"}_ 

[image]

Can you help me? Am I doing something wrong? 

Thank you a lot

shaman171, 2022-11-29:
OK I found out problem -> I had in _configuration.yam_l old modbus integration defined -> when I deleted it -> it seems it is working -> I was able to set up the integration.

sfstar, 2022-11-30:
Thank you for taking the time to test the integration (it helps a lot with detecting issues that I might miss in my testing).

Yeah the error given if pymodbus < 3.0.0 is used is sadly really generic.
Not using integrations that use 2.x.x versions of pymodbus solves your issue (for example the solaredge modbus tcp integration latest version will work with this integration enabled while older versions won't).

When the integration gets stable enough I will look into getting the core modbus integration up to 3.x.x of the pymodbus package.
So that users can add this integration in conjunction with their existing modbustcp integration setup to allow for a less breaking switchover.

Will leave this issue open for now so that anyone having the same problem can more easily find the solution

aheadley, 2022-12-01:
Hi, I'm ran into the same "`Invalid handler specified`" error when I tried to set up this integration. HA is running from the docker (actually podman) container if that might make a difference. There's also this line in the log:
`2022-12-01 00:06:55.909 ERROR (MainThread) [homeassistant.config_entries] Error occurred loading configuration flow for integration victron: cannot import name 'ModbusTcpClient' from 'pymodbus.client' (/usr/local/lib/python3.10/site-packages/pymodbus/client/__init__.py)`

As far as I know, I'm not using any integrations that would have conflicted over pymodbus, so I don't know what was causing it. I've worked around it by force upgrading pymodbus in the container to 3.0.2 for now. In case anyone else ran into this, here is what I did (restart HA afterwards):
```
$ podman exec -it <HA container name> /bin/bash
bash-5.1# /usr/local/bin/pip3.10 install pymodbus==3.0.2
Collecting pymodbus==3.0.2
  Using cached pymodbus-3.0.2-py2.py3-none-any.whl (142 kB)
Installing collected packages: pymodbus
  Attempting uninstall: pymodbus
    Found existing installation: pymodbus 2.5.3
    Uninstalling pymodbus-2.5.3:
      Successfully uninstalled pymodbus-2.5.3
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
pysmarty 0.8 requires pymodbus==1.5.2, but you have pymodbus 3.0.2 which is incompatible.
Successfully installed pymodbus-3.0.2
```

sfstar, 2022-12-02:
@aheadley thanks for contributing a work-around for people running their own dockerized HA
So to update on the co-existence issue between the modbus integration and this integration:
I submitted a PR to update the core version but it seems that the maintainer of the integration wants to wait for the 3.1.x release before updating.
Expected timeline for the pymodbus 3.1.0 version is somewhere in december.
So it would be a reasonable expectation that around january of 2023 this and the built in modbus integration will be able to co-exist.

sticilface, 2022-12-11:
That is a shame as I have a need for both the victron and the existing modbus set up.  Any workaround to get them to both work?

sfstar, 2022-12-12:
Not in the same home assistant instance.
Since, there can only be one pymodbus library version installed per homeassistant core runtime environment.
You could try running 2 homeassistant instances and link them together with the remote homeassistant integration "https://github.com/custom-components/remote_homeassistant".
Otherwise the only option is to wait until the integrated modbus integration also starts using a v3.x.x pymodbus version.

trentjw, 2022-12-19:
got this error as well `Config flow could not be loaded: {"message":"Invalid handler specified"}` from what I gather on the comments this is not really ready to be intergrated into home assistant well I guess I will stick to my old method of build the sensors by hand in modbus

sfstar, 2022-12-20:
@trentjw to clarify on your statement:
The integration is ready to be integrated into home assistant.
Currently it's just a choice between the build-in modbus integration with yaml configuration and this integration(/multiple other custom modbus integrations like: [solaredge multi modbus](https://github.com/WillCodeForCats/solaredge-modbus-multi)) that have already moved to the pymodbus > 3.x.x release.

The core modbus integrations maintainer stated that the core modbus integration will also update to 3.x.x once the pymodbus 3.1.x release has been made.
This is planned for this month so in a month or two the core modbus integration will be able to be run again in parallel with this integration.

sfstar, 2023-02-05:
Closing this issue as from 2023.2.0 and onwards home assistant core modbus integration uses pymodbus 3.1.1.

closed 2023-02-05 by sfstar