sfstar/hass-victron #2: Phases from grid are mixed

shaman171, 2022-11-30:
Hello,

I have been testing this further and found out, that phases from grid are mixed for some reason. 

[image]

It seems that it is like: 
Victron  -  Homeassistant
L1  -  L2
L2  -  L3
L3  -  L1

It can be easy changed in Homeassistant with Template sensors, but is there way to change it natively? Or it cannot be somehow controlled and it base on the discovering of devices? 

Thank you a lot for help and wish you nice day

sfstar, 2022-11-30:
Thank you for submitting this potential bug.
I have 2 Gavazzi et112 to test the integration with only a single grid phase. (So having feedback on 3 phase use of the integration is really helpfull)

I did notice that my output phases seem to do something similair.
Although that behaviour was observed while the AC out of my multiplus wasn't yet connected to anything. (Still building the physicall system while creating this integration).
Therefore I did discard my observation as a false positive (since I only observed this on the consumption side with no loads or even a complete circuit attached.)

Could you provide me with the following information to further narrow down this issue?
- What is selected under "settings -> ESS -> Grid metering" (is this "inverter/charger" or external meter)?
- (If external meter is used) What is the model (i.e. et112, em24) of the (gavazzi) meter you are using?
- Are/were the values reported correctly if you used the built-in modbustcp integration configuration to read those register addresses?

The register addressing dict for the system L registers seem to be correctly defined in the integration itself
```
    "system_grid_L1": RegisterInfo(820, INT16, UnitOfPower.WATT),
    "system_grid_L2": RegisterInfo(821, INT16, UnitOfPower.WATT),
    "system_grid_L3": RegisterInfo(822, INT16, UnitOfPower.WATT),
```

If you compare it to a copy of the same register values from the victron modbus tcp definition spread sheet
```
com.victronenergy.system | Grid L1 | 820 | int16 | 1 | -32768 to 32767 | /Ac/Grid/L1/Power | no | W | Power supplied by Grid to system.
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
com.victronenergy.system | Grid L2 | 821 | int16 | 1 | -32768 to 32767 | /Ac/Grid/L2/Power | no | W |  
com.victronenergy.system | Grid L3 | 822 | int16 | 1 | -32768 to 32767 | /Ac/Grid/L3/Power | no | W |  
```

It could be that there is a small error in the register id's in the documentation that causes the incorrect L reporting.
However, in that case the modbus tcp integration (with yaml configuration) should result in the same behaviour.
If the modbus yaml integration does report the correct system L values then it could be that there is a slight error in my (this integrations) data parsing.

Looking forward to your reply

shaman171, 2022-12-01:
Hello, 

I am happy to test it. Already like first issue it works great and make Victron integration to Homeassistant much easier! 

I am also not yet fully done with my system, but I will try to give as much input as I can to help to solve the issue. 

> What is selected under "settings -> ESS -> Grid metering" (is this "inverter/charger" or external meter)?

For now still inverter / charger.

> (If external meter is used) What is the model (i.e. et112, em24) of the (gavazzi) meter you are using?

For now I do not have it installed yet, but in near future it will be ET340 

> Are/were the values reported correctly if you used the built-in modbustcp integration configuration to read those register addresses?

I am not fully sure how I can test this one? I am just a bit skilled user of Homeassistant. If you would not mind to tell me more hints how to test this please? :) 

Thank you a lot again!

sfstar, 2022-12-02:
Thanks for your reply.
One small note:
The inverter/charger setting uses the built-in estimation of your inverter.
This is known to not be accurate in displaying actual power going in to and out from the device.

Having said that the integration should still report the measurements as reported by victron (and therefore displayed in the GX UI).

When you install the ET340 I will be really eager to know if there are any issues with this integration and that meter type.
For now you could provide me with more information by using this script (modified version of the one provided by @dummy74 in #3)

```
from pymodbus.constants import Defaults
from pymodbus.constants import Endian
from pymodbus.client.tcp import ModbusTcpClient as ModbusClient
from pymodbus.payload import BinaryPayloadDecoder

Defaults.Timeout = 25
Defaults.Retries = 5
client = ModbusClient('<change to your ip>', port='502')

for reg in [820,821,822]:
	result = client.read_input_registers(reg,2)
	decoder = BinaryPayloadDecoder.fromRegisters(result.registers, byteorder=Endian.Big)
	r1=decoder.decode_16bit_int()
	print(f"Register {reg} : {r1}")
```
Please note that you need to set your own ip address where `<change to your ip>` is currently stated.
You can then run is from a computer with the pymodbus package and python >= 3.9 installed.

It was a bit easier for me to recycle the script provided by dummy74.
However, if you don't have an python3 dev env ready and/or can't create one, I will be posting the instructions for reading the registers with the modbus tcp service of home assistant during this weekend.
It needs a lot more explaining / screenshots to get everything to work (mostly due to the 2.5.3 version conflicting with the pymodbus version of this integration). 
Thereby needing additional instructional steps on enabling and disabling the integrations without running into issues.

sfstar, 2022-12-02:
I believe the solution for #3 (see that issue for further description on what I fixed) could also have fixed your issue.
Since the grid values being:
```
system_consumption_L3 <-> system_grid_L1
system_grid_L1 <-> system_grid_L2
system_grid_L2 <-> system_grid_L3
```
Could also explain your observed behaviour.
If the latest version resolves your issues (and if it doesn't) please let me know

shaman171, 2022-12-02:
Hello, 

I do not have now time for more testing with those scripts you wrote above - I can do later. But I downloaded the latest revision you just issued -> and all work great! 

[image]

[image]

It is really awesome working now! 👍 

Thank you a lot!

sfstar, 2022-12-02:
Hello @shaman171 ,

In that case you don't have to do the tests with the script I posted above.
I will close the issue as resolved now.
Feel free to re-open if there are complications

closed 2022-12-02 by sfstar