sfstar/hass-victron #3: Phases from system consumption / genset not correct

dummy74, 2022-12-02:
Hello,
I have installed your integration today. Unfortunately, something seems to be mixed up for me as well. At the "system consumtion" the value for L1 seems to be at L2, L2 at L3 and L3 is at "system genset L1".  There is no generator for me!

I have also checked the whole thing again via the MQTT interface using MQTT Explorer.

Nevertheless: This is a very nice integration. Now only the small errors have to be removed.

PS: I have checked the register definition. This seems to be correct.

Many greetings
Holger
[image]
[image]

dummy74, 2022-12-02:
@sfstar: When I read the registers directly from Python the values are correct. The register definition on the part of Victron is valid!

```
from pymodbus.constants import Defaults
from pymodbus.constants import Endian
from pymodbus.client.tcp import ModbusTcpClient as ModbusClient
from pymodbus.payload import BinaryPayloadDecoder

Defaults.Timeout = 25
Defaults.Retries = 5
client = ModbusClient('192.168.2.56', port='502')

for reg in [817,818,819]:
	result = client.read_input_registers(reg,2)
	decoder = BinaryPayloadDecoder.fromRegisters(result.registers, byteorder=Endian.Big)
	r1=decoder.decode_16bit_uint()
	print(f"Register {reg} : {r1}")
```
Output:
```
Register 817 : 172
Register 818 : 38
Register 819 : 69
```

sfstar, 2022-12-02:
Thank you for opening this issue.
Most likely the integration is doing something wrong when decoding the bytebuffer received from the GX device.

After a quick investigation it seems that a similair issue is locally reproducable with my setup.
Although the interesting thing is that for me it seems to assign the grid L1 values to the grid L2 entity
[image]

While your script returns the correct values for my setup as well

```
Register 817 : 22
Register 818 : 0
Register 819 : 0
```

I need to further investigate before an solution can be introduced.
I think I will have some time this weekend to further troubleshoot

sfstar, 2022-12-02:
So I had a sudden insight in what it could be.
It turned out that the register names were broken for pvonoutput for the system registers.

```
    "system_pvonoutput_L1": RegisterInfo(808, UINT16, UnitOfPower.WATT),
    "system_pvonoutput_l1": RegisterInfo(809, UINT16, UnitOfPower.WATT),
    "system_pvonoutput_l1": RegisterInfo(810, UINT16, UnitOfPower.WATT),
```
This caused the offset by one due to the integration using an ordered dict (with unique key requirement) for parsing the data.
So 808 and 810 would be decoded but 809 wouldn't since it wasn't an unique dictionary key (808 had L1 in the name and 809/810 l1 in lower case)
By changing this to

```
    "system_pvonoutput_L1": RegisterInfo(808, UINT16, UnitOfPower.WATT),
    "system_pvonoutput_L2": RegisterInfo(809, UINT16, UnitOfPower.WATT),
    "system_pvonoutput_L3": RegisterInfo(810, UINT16, UnitOfPower.WATT),
```
The system registers were reporting the correct values for my local setup.
@dummy74 could you test it with your setup as well?

Fun fact: during the fixing I also discovered that I was decoding INT32 as UINT32. So that was also resolved.

dummy74, 2022-12-02:
Perfect! The values are now displayed correctly.

sfstar, 2022-12-02:
@dummy74 thank you for confirming.
I will close this issue as resolved now.
Please feel free to re-open if there are complications

closed 2022-12-02 by sfstar