sfstar/hass-victron #4: Sensors not available on energy dashboard

FcoCalero, 2022-12-04:
Hi, 

Would be nice to have the sensors avaiable for the energy dashboard. Thanks for this integration :) 

Regards

sfstar, 2022-12-05:
Hello @FcoCalero,

Do you mean the home assistant built in energy dashboard? 
I.e. this:
[image]

In that case the sensors should be available in the energy dashboard configuration section.
I.e. this page:
[image]

You can go to this page by clicking the "my energy configuration" button below:
[image]

If this is not what you meant please let me know.

FcoCalero, 2022-12-05:
Yes, is what i meant. I've just done a clean installation to check it a second time and i confirm they are not available in the energy dashboard configuration section after provide the GX ip.

sfstar, 2022-12-06:
Hmm strange.
This is what I see when I try to add another battery system:
[image]

It seems that I can't recreate the observed behaviour on my end.
Could you share your local integrations page (namely the victron entry and listed entities):
(should look like this):
[image]

And also share a screenshot of you trying to add a specific entity in the energy configuration dashboard (to see what entities are available to you).

Thanks in advance

FcoCalero, 2022-12-06:
Sure no problem.

[image]
[image]

FcoCalero, 2022-12-07:
Here is the equipment i have working.

[image]
[image]

Also i paste here the sensors i've been using in HA out of this integration. Another thing i've noticed is that for example, Grid lost alarm and Battery State of Charge are not listed but that's for another thread maybe. 

      - name: "Grid Power"
        scan_interval: 10
        slave: 100
        address: 820
        data_type: int16
        unique_id: Victron_grid_power
        unit_of_measurement: 'Wh'
        state_class: 'total_increasing'
        device_class: 'energy'
         
      - name: "AC Loads"
        scan_interval: 10
        unit_of_measurement: 'kWh'
        slave: 100
        address: 817
        data_type: int16
        unique_id: Victron_ac_loads
        state_class: 'total_increasing'
        device_class: 'energy'
      
      - name: "Inverter State"
        scan_interval: 10
        slave: 227
        address: 31 
        data_type: int16
        unique_id: Victron_inverter_state
        
      - name: "ESS State"
        scan_interval: 10
        slave: 100
        address: 2900
        data_type: int16
        unique_id: Victron_ess_state
        
      - name: 'Grid lost alarm'
        slave: 227
        address: 64
        data_type: uint16
        unit_of_measurement: "0=Ok;1=Warning;2=Alarm"

      - name: "Battery State of Charge"
        scan_interval: 10
        slave: 100
        address: 843
        data_type: int16
        unique_id: Victron_ess_battery

      - name: "Battery status"
        unit_of_measurement: 'kWh'
        scan_interval: 10
        slave: 100
        address: 844
        data_type: int16
        unique_id: Victron_battery_status
        state_class: 'measurement'
        device_class: 'energy'
        
      - name: "Charger power"
        scan_interval: 10
        unit_of_measurement: 'kWh'
        slave: 100
        address: 866
        data_type: int16
        unique_id: Victron_charger_power
        state_class: 'total_increasing'
        device_class: 'energy'

sfstar, 2022-12-07:
So I might have found the cause for your issue.
Your screenshots show that you are running the following firmware versions:
GX: v2.84
multiplus: 481

While i'm running:
GX: v2.92
multiplus: 492

Where this might lead to the issues you are observing is that the victron modbus TCP spec sheet states that rev 43 introduces:
```
  | Add registers for VE.Bus Multi energy counters |  
  | Add registers for wakespeed alternators |  
  | Fix incorrect unit for PV-Inverter powerlimit register |  
  | Add new tank types to fluid type enum |  
  | Add additional diagnostic items for CAN-bms batteries |  
  | Add LowSoc alarm register for Inverter RS and Multi RS |  
  | Fix remarks for register 2903 |  
```

Specifically:
```
  | Add registers for VE.Bus Multi energy counters |  
```
The integration treats all vebus registers as one request.
Including:
```
    "vebus_acin1toacout": RegisterInfo(74, UINT32, UnitOfEnergy.KILO_WATT_HOUR, 100),
    "vebus_acin1toinverter": RegisterInfo(76, UINT32, UnitOfEnergy.KILO_WATT_HOUR, 100),
    "vebus_acin2toacout": RegisterInfo(78, UINT32, UnitOfEnergy.KILO_WATT_HOUR, 100),
    "vebus_acin2toinverter": RegisterInfo(80, UINT32, UnitOfEnergy.KILO_WATT_HOUR, 100),
    "vebus_acouttoacin1": RegisterInfo(82, UINT32, UnitOfEnergy.KILO_WATT_HOUR, 100),
    "vebus_acouttoacin2": RegisterInfo(84, UINT32, UnitOfEnergy.KILO_WATT_HOUR, 100),
    "vebus_invertertoacin1": RegisterInfo(86, UINT32, UnitOfEnergy.KILO_WATT_HOUR, 100),
    "vebus_invertertoacin2": RegisterInfo(88, UINT32, UnitOfEnergy.KILO_WATT_HOUR, 100),
    "vebus_invertertoacout": RegisterInfo(90, UINT32, UnitOfEnergy.KILO_WATT_HOUR, 100),
    "vebus_outtoinverter": RegisterInfo(92, UINT32, UnitOfEnergy.KILO_WATT_HOUR, 100)
```

It might be the case that your firmware version doesn't contain these register entries.
Which would cause the scan request for a vebus device on unit address 227 to fail.
Hence these registers not being available as entities to you.
I will look into creating a seperate list so that an older vebus model registers being present while the above mentioned registers are not won't cause the entire vebus device to be skipped.

If this fixes your issue it won't mean that all older firmware devices will work without any issues.
Since victron has made more changes and revisions and it isn't that clear what fw matches which modbus tcp spec revision.
What makes this (supporting older firmware versions) even more difficult is that it isn't always clear based on the revision information which registers were present before and which were added after.

Victron added register 74 until 92 with information as to the total energy going from/to the acin1/acout and battery.
Since you might want to use the information these register provide in your energy dashboard I recommend updating to the latest victron firmware version for the best experience with this integration.

Note: Splitting the integrations register list for vebus into 2 lists will require all users with an vebus device to rescan for available devices.
I'm hoping to find an solution that allows me to force a rescan upon an upgrade but if that can't be done it might take a while for this change to be included in an integration update. (In case this is not possible I would like to group breaking changes in order to limit the impact of an upgrade on existing users.)

FcoCalero, 2022-12-07:
ok, makes sense, thanks. In my case i think im gonna try to upgrade the firmware and see the differencies, hope for better :). I will let you know the behaviour once the new firmware is working.

About the issue to scan other devices, what if you add a text box during the intial configuration to look for devices ID manually.

FcoCalero, 2022-12-07:
After GX device firmware updated, this is how it looks like.

[image]

After reboot HA

[image]

Now i have added the entities which i saw they are incrementing

[image]

Thank so much :)