sfstar/hass-victron #5: Outbound Connection, Why?

tweak19, 2022-12-07:
Hi,
as you recommend in the haos forum I tried your addon without hacs.
Why there are any outbound connections required for this addon?
I can see new connections after enable this integration.

Thanks for Feedback!

[image]

sfstar, 2022-12-07:
Hello @tweak19,

Thank you for opening this issue.
As far as I know the integration (and the pymodbus library it uses) should only open a connection to your GX device.
Could you provide me with an example of the connections that you are seeing (wireshark screenshot or pcap file etc)?

This could provide me with a clue as to what is happening and whether it is the integration that initiates these calls or a mechanic in home assistant itself.

sfstar, 2022-12-08:
No problem 😉.
Hope you will enjoy using the integration.
I will close this issue as it seems to be resolved.
Feel free to (re-)open an issue in the future.

closed 2022-12-08 by sfstar