sfstar/hass-victron #6: Duplicate entities without value updates

dummy74, 2022-12-08:
I happened to notice that on one of the devices ('solarcharger'), some entities (trackers) are no longer updating in the device settings. On the other hand, the entities with the same name in the 'settings' device are updated. If you look at the entities under 'Settings' -> 'Devices and Services' -> 'Entities' you see that there are duplicate entries!

sfstar, 2022-12-08:
Hello @dummy74.
Thank you for opening this issue.
Could you provide me with:
- The entity id's of the working 'settings' device tracker entities
- The entity id's that are no longer updating (of the 'solarcharger') device
- A screenshot of the duplicate entries you see under ('Settings' -> 'Devices and Services' -> 'Entities' )
- The unit ids of your solarcharger(s) in the gx device ('menu' -> 'settings' -> 'services' ->  'Modbus TCP' -> 'available services')

sfstar, 2022-12-09:
Thank you for submitting the screenshots.
Based on what I'm seeing I think this issue might be caused by multiple solartrackers being present in the system (I haven't tested for this use case).

Could you confirm whether you have multiple solarcharger devices connected to your setup?
Furthermore, could you add the following to your yaml config:
```
logger:
  default: info
  logs:
    custom_components.victron: debug
```
This will enable debug logging for the integration after a restart.
If you have setup my homeassistant you can click this button in order to navigate to the logs:
[image]

Otherwise you go to 'settings' -> 'system' -> 'logs'

Once you are on the logs page you should click "Load full logs" (or download all logs and submit them instead).
And look for the following lines:
```
2022-12-09 06:37:52.370 DEBUG (MainThread) [custom_components.victron.coordinator] Fetching victron data
2022-12-09 06:37:52.370 DEBUG (MainThread) [custom_components.victron.coordinator] {30: ['gavazi_grid_registers'], 31: ['acload_registers'], 100: ['settings_registers', 'settings_ess_registers', 'system_registers', 'system_battery_registers', 'system_dc_registers', 'system_charger_registers', 'system_power_registers', 'system_bus_registers'], 228: ['vebus_registers']}
2022-12-09 06:37:52.464 DEBUG (MainThread) [custom_components.victron.coordinator] Finished fetching victron data in 0.094 seconds (success: True)
```

These loglines will be generated everytime the integration updates the entities (in other words it matches the interval config option).
Please add your loglines to the reply as well.
This will allow me to determine what units were detected and which register lists were assigned to each unit.

Lastly please add screenshots of your GX device page 'menu' -> 'settings' -> 'services' -> 'modbus TCP' -> 'available services' that show all connected devices

sfstar, 2022-12-10:
Thank you for providing the requested information.

On a first glance it seems that the solarcharger settings of the ve.can 0 device seem to get listed under the system ID (100)
Based on the following screenshot (from the victron modbus TCP spreadsheet) this seems to be a correct assumption:
[image]

You state that only one of the MPPT trackers is currently connected to solar modules.
This could explain why only tracker values of one of the solarchargers is being updated.
Since there wouldn't be any new data for the solarcharger that doesn't have any solar modules connected.

As to the duplicate naming.
The integration publishes unique id's and entity names for all devices.
The entity naming for the can bms device and the system settings (unit 100 for system and 225 for can bms respectively) is the register name as is.

While for all other units the entity name is the register name + '_' + unit id
In your case the unit 0 is the first ve.can device in your system.
Victron publishes all data for unit 0 under the system unit id of 100 (this is important later).
While unit 1 is your 2nd ve.can device in your system.

Based on these rules the entity naming you are seeing is correct / not an homeassistant internal duplication fix.
Although I see why the _1 could give that impression.

Since the unit 1 doesn't have any solar modules installed it stands to reason there will be no data updates until there is new data available (in other words PV has been installed on at least 1 tracker of that solarcharger device).
While the unit 0 (your pv connected solarcharger) does update but is currently being grouped under the system 'device'.

In order to reduce confusion for new users i'm going to be making a release that will only group system and settings registers under unit 100.
This will also mean that unit 100 non system/settings entities will get an new entity id (only available after a reinstall of the integration).
This new entity id will be the register name + "_" + 0 (this will ensure that the ve.can unit 0 matches the general entity id naming scheme).

The release will also group the ve.can 0 device entities into a seperate (not the system/settings) device.

sfstar, 2022-12-10:
@dummy74 Since I can't test this change could you install the beta version: `v0.0.5-beta.1` and test the change?
Thanks in advance

sfstar, 2022-12-10:
> I thought until now that the 4 trackers results from my hardware configuration with the 2 MPPT RS 450/100-TR with 2 trackers each.

Yeah the trackers are 0 to 3 for every solarcharger detected.
Victron just pre-emptively added registers for tracker 2 and 3 possibly because they believe they might just release a 4 tracker model somewhere in the future.

dummy74, 2022-12-10:
> @dummy74 Since I can't test this change could you install the beta version: `v0.0.5-beta.1` and test the change? Thanks in advance

I just installed your update. Now I have an additional device (2x 'solarcharger'). I removed the now invalid entities and replaced them with the new entities in my dashboards (appending '_0'). As far as I can see everything is working now. The restructuring definitely makes the underlying system structure clearer. Great!!!

dummy74, 2022-12-10:
> Victron just pre-emptively added registers for tracker 2 and 3 possibly because they believe they might just release a 4 tracker model somewhere in the future.

There are already solar charge controllers with 4 trackers -> 'MPPT RS 450/200-TR'.

closed 2022-12-10 by dummy74

sfstar, 2022-12-11:
Ah didn't know that. 
Great to hear everything is working ok at your end.
I will promote the release to an official release today