sfstar/hass-victron #7: Power "W" values of by 10x!

firefox7518, 2022-12-08:
Hi

I just installed your integration and I noticed on the values that a lot of the power "W" values are 10time less than they should be.
Look at my screenshot below.
[image]

As you see from an Ampere point of view the Wattage values are off 10x. L1 loaded with 2,7A at 240V cannot be 43W. It is actually 430W. Same applies to L2 Power, L3 Power etc. 

Where can I correct this?

sfstar, 2022-12-08:
Hello and thank you for opening this issue.
The scale decoding that the integration applies is based on victrons modbus tcp ledger:
```
com.victronenergy.vebus | Output power 1 | 23 | int16 | 0 | -327680 to 327670 | /Ac/Out/L1/P | no | VA or Watts
-- | -- | -- | -- | -- | -- | -- | -- | --
com.victronenergy.vebus | Output power 2 | 24 | int16 | 0 | -327680 to 327670 | /Ac/Out/L2/P | no | VA or Watts
com.victronenergy.vebus | Output power 3 | 25 | int16 | 0 | -327680 to 327670 | /Ac/Out/L3/P | no | VA or Watts
com.victronenergy.vebus | Battery voltage | 26 | uint16 | 100 | 0 to 655.35 | /Dc/0/Voltage | no | V DC
```
As can be seen above it states a scalefactor of 0 for those entities.
Which is also set in the integration translation of this ledger for those registers:
```
    "vebus_out_L1_power": RegisterInfo(23, INT16, UnitOfPower.WATT, 0),
    "vebus_out_L2_power": RegisterInfo(24, INT16, UnitOfPower.WATT, 0),
    "vebus_out_L3_power": RegisterInfo(25, INT16, UnitOfPower.WATT, 0),
    "vebus_battery_voltage": RegisterInfo(26, UINT16, ELECTRIC_POTENTIAL_VOLT, 100),
```

You are most likely running into a bug I assumed might occur due to the integrations interpretation of the scaling factor.
for 0 I assumed that victron meant return as is.
But based on your (very detailed - thank you) provided information the correct decoding of 0 should be * 10.
Or in other words literally add an zero to the returned number if 0 is the scaling factor.

I will be introducing a patch shortly to address this issue.
Please wait up to 4 hours for the patch to become available in hacs

sfstar, 2022-12-08:
Version v0.0.3 has just been released containing the fix for this issue.
Feel free to let me know if it resolves the issue for you as well.

firefox7518, 2022-12-09:
HI sfstar

Thanks for trying to fix it. However just using *10 will not make it actually correct. Its off by several Watts this way.

I checked the Victron Modbus docs and in the Excel Spreadsheet they clearly state Scalefactor of 0.1 as the values can be -327680 to +327670.
[image]

Your fix is therefore not fully correct and leaves a good bit of precision behind.

sfstar, 2022-12-09:
Hello firefox7518
 
Thanks for your feedback
I redownloaded a fresh copy of the modbus TCP spreadsheet and that copy shows the following:
[image]

Which differs from your spreadsheet screenshot.
The version number in my spreadsheet title states 2.90.
What is the version of the document you are using?
(Just trying to find out if there is a newer version available that might have even more differences to my reference sheet)

sfstar, 2022-12-09:
Disregard my latest message.
I was opening the sheet in numbers on my mac.
Opening the same vanilla file in ondedrive online excel does display the 0.x values.
There aren't even any scalefactors of 0 present in the reference it turns out.

Most likely this is caused by the method of noting 0 decimal numbers (this `0.1` vs `0,1`)
Will look into creating a patch that takes the correct scalefactor into account (that also updates the incorrect ledgers in the const.py file)

sfstar, 2022-12-09:
@firefox7518 I have published a new release with some changes regarding the scalefactor.
Could you test whether the latest release version resolves the output register issues?
(My system is currently non operational due to physical changes I'm making to my fusing, the change was briefly validated against demo mode)

firefox7518, 2022-12-09:
Hi

Here a screenshot of the cerbo real-time values and the ones shown in HA.
[image]

As you can see the values are off always. It's like rounding up or down. I did not check your code if you are doing it but it would be strange if the modbus values differ from the onscreen or MQTT values.

sfstar, 2022-12-10:
Sorry for my late reply.
Thank you for testing and providing the screenshot.

Currently my setup has been reassembled and i've tested what the modbus tcp server returns.
I've done this via the built-in modbus integration in order to rule out if this integration is causing these rounding errors.

used config:
```
modbus:
  - name: victron
    type: tcp
    host: <ip>
    port: 502
    retry_on_empty: true
    retries: 10
    close_comm_on_error: true
    sensors:
      - name: victron_vebus_out_power
        slave: 228
        unit_of_measurement: W
        address: 23
        data_type: int16
        scale: 10
        scan_interval: 1
```
One thing to note is that the `scale: 10` equals the scale 0.1 from this integrations perspective.
Another thing to note is that setting no scale just returns the value (example) 504 as 50.

As can be seen in the following screenshots (for these screenshots both the GX UI and the modbus integration has a refresh rate of 1 second).
1.
[image]
2.
[image]
3.
[image]

Based on these screenshots the following can be stated:
The modbus tcp server (GX device)rounds the value before sending it to any readers.

Why victron chose to round instead of just sending the value as is (like for the MQTT integration) I don't know.
It doesn't make any sense in my eyes as (even if the chosen datatype of int16 is to small for large installations) victron could have just used int32 registers instead.

Personally I can live with the rounding done by the modbus server. (It doesn't interfere with basic operations)
Although having the raw values instead of pre-rounded ones would have been preferred.

As far as i can see the issue with incorrect scaling by this integration has been resolved.
Please let me know if you see it the same way (or not) and whether this issue can be closed.

firefox7518, 2022-12-10:
Hi
No problem about the response time. Everything good. I expected you answer as I also did some testing use modbus poll to gather some data. Now that we know that your code is flawless in this regard it is of course fine to close this issue. 

Thanks a lot for the fast fixing. I did not check all the values where 0.1 was the scale factor but I guess your fix was for all of them.

You can close the issue, thanks a lot and best wishes from Switzerland

sfstar, 2022-12-10:
No problem.
I made sure to triple check when I replaced the 0 scales with the actual 0.1 and 0.01 values they were suppossed to be.
It helps alot that I'm able to actually see the scalefactor victron intended 😉.

Good luck to you too.

closed 2022-12-10 by sfstar