sfstar/hass-victron #8: Missing VE-Can MPPT 250/85

mmaritz, 2022-12-11:
Wonderful integration - many thanks.

I am missing my VE.Can MPPT when the integration discovers the modbus devices.  
Latest upgrade done as well.  

Regards
Mark

sfstar, 2022-12-11:
Thank you for opening this issue.
The integration scans only the unit id's that are present in the victron modbus spec.
unit id 280 isn't currently defined in this document.
Adding your device as a scanned for unit id could result in a working setup but it could also reveal other currently unknown issues.
Since it is not known what else has been changed in the release candidate version of v3.00-8.

I will make beta release for you to use with a rescan to find unit 280.
However, until victron releases the updated modbus TCP spec for the v3.00 release I won't be making a effort to fix any additional release candidate specific issues.
Since, it will take a lot of time to reverse engineer issues that arise from the spec change between releases.

As the saying goes: "Here be dragons"

sfstar, 2022-12-11:
You can install the beta release: `v0.1.1-beta.1` to see whether your solar charger is correctly detected after a rescan (or you can delete and add the integration to the same effect)

mmaritz, 2022-12-11:
Thanks for the feedback!

mmaritz, 2022-12-11:
Sadly no luck...  upgraded and also deleted and added.

sfstar, 2022-12-12:
Thank you for trying the beta out.
In that case it is really likely that victron has also changed the modbus register addresses of the solarcharger devices in the release candidate.
For now you could try to downgrade to the latest stable gateway release (if at all possible).

sfstar, 2022-12-12:
The won't fix label will remain until the v3 release is stable and the register documentation is available.
After which the integration will be updated to fix your issue.

mmaritz, 2022-12-12:
Thanks @sfstar 

Done the downgrade to 2.92 but no luck so will wait for the stable V3.00 release and updated Modbus Register List

OlePZ9, 2022-12-23:
I think it is a problem with mapped devices, that are not well documented in the xls file. I was looking like hell to find out the right parameters to get the data from my 250/60
[image]

sfstar, 2022-12-23:
Weird, since 226 is a unit id that the integration scans for (including the 789 register address).
So if that modbus yaml configuration is working the integration should also pickup this device.
Are you running the 3.0.0 pre release as well?

OlePZ9, 2022-12-23:
Yes, this yaml config is working well. Actually I am running firmware 2.92

mmaritz, 2022-12-24:
Sorry to cross post from another forum but this will help with more info on the issue:
[https://energytalk.co.za/t/new-modbus-registers/2744](https://energytalk.co.za/t/new-modbus-registers/2744)

sfstar, 2022-12-24:
Great find @mmaritz, I personally had a similair issue with duplicate modbus ID's after changing a ET112 role in the system.
Fixed it myself by editing the /data/conf/settings.xml file, since I was setting up a new system and old data isn't important to me.

sfstar, 2024-03-15:
According to the new modbus register list provided by victron:
[image]

The mppt should be translated to respond on query's on unit id 215.
Therefore, this issue might be fixed in the next release.

sfstar, 2024-03-29:
The release (v0.2.0) has been shipped.
Closing issue, feel free to re-open if the issue persists with the new release

closed 2024-03-29 by sfstar