sfstar/hass-victron #9: Grid Meter and battery stops getting data

braamvh, 2022-12-11:
I have started using this integration today only, so I don't have a lot of experience with it yet.

My ET112 grid meter (unit ID 30) have stopped providing data after some time. Is there a way to debug the cause of this? Other sensors seems to work correctly so the CCGX responding correctly. I did a full rescan as well, just in case that's the issue. My CCGX and the VRM portal both show the grid meter as present. My previous modbus configuration did also work correctly.

The same is happening to the battery (ID 247).

[image]
[image]

At the time the data stopped, I got these messages (once off) in my HA logs. I've trimmed off some at the end to keep the comment short enough to post.

```
2022-12-11 20:04:57.247 WARNING (MainThread) [custom_components.victron.coordinator] no valid data returned for entities of slave: 30 if device was physically removed please force a rescan to resolve this issue
2022-12-11 20:04:57.399 WARNING (MainThread) [custom_components.victron.coordinator] no valid data returned for entities of slave: 247 if device was physically removed please force a rescan to resolve this issue
2022-12-11 20:04:57.405 WARNING (MainThread) [custom_components.victron.coordinator] no valid data returned for entities of slave: 247 if device was physically removed please force a rescan to resolve this issue
2022-12-11 20:04:57.527 ERROR (MainThread) [homeassistant.helpers.entity] Update for sensor.victron_battery_alarm_hightemperature_247 fails
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 527, in async_update_ha_state
    await self.async_device_update()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 722, in async_device_update
    raise exc
  File "/config/custom_components/victron/sensor.py", line 153, in async_update
    data = self.description.value_fn(self.coordinator.processed_data(), self.description.slave, self.description.key)
  File "/config/custom_components/victron/base.py", line 10, in <lambda>
    value_fn: Callable[[dict], StateType] = lambda data, slave, key: data["data"][str(slave) + "." + str(key)]
KeyError: '247.battery_alarm_hightemperature'
2022-12-11 20:04:57.532 ERROR (MainThread) [homeassistant.helpers.entity] Update for sensor.victron_battery_alarm_midvoltage_247 fails
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 527, in async_update_ha_state
    await self.async_device_update()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 722, in async_device_update
    raise exc
  File "/config/custom_components/victron/sensor.py", line 153, in async_update
    data = self.description.value_fn(self.coordinator.processed_data(), self.description.slave, self.description.key)
  File "/config/custom_components/victron/base.py", line 10, in <lambda>
    value_fn: Callable[[dict], StateType] = lambda data, slave, key: data["data"][str(slave) + "." + str(key)]
KeyError: '247.battery_alarm_midvoltage'
2022-12-11 20:04:57.535 ERROR (MainThread) [homeassistant.helpers.entity] Update for sensor.victron_battery_alarm_lowfusedvoltage_247 fails
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 527, in async_update_ha_state
    await self.async_device_update()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 722, in async_device_update
    raise exc
  File "/config/custom_components/victron/sensor.py", line 153, in async_update
    data = self.description.value_fn(self.coordinator.processed_data(), self.description.slave, self.description.key)
  File "/config/custom_components/victron/base.py", line 10, in <lambda>
    value_fn: Callable[[dict], StateType] = lambda data, slave, key: data["data"][str(slave) + "." + str(key)]
KeyError: '247.battery_alarm_lowfusedvoltage'
2022-12-11 20:04:57.537 ERROR (MainThread) [homeassistant.helpers.entity] Update for sensor.victron_battery_alarm_highfusedvoltage_247 fails
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 527, in async_update_ha_state
    await self.async_device_update()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 722, in async_device_update
    raise exc
  File "/config/custom_components/victron/sensor.py", line 153, in async_update
    data = self.description.value_fn(self.coordinator.processed_data(), self.description.slave, self.description.key)
  File "/config/custom_components/victron/base.py", line 10, in <lambda>
    value_fn: Callable[[dict], StateType] = lambda data, slave, key: data["data"][str(slave) + "." + str(key)]
```

braamvh, 2022-12-11:
Strange. An hour or so later it started working again on its own and populating all the entitiies. I'll keep an eye on it and see if this happens again.

sfstar, 2022-12-12:
Hello and thank you for opening this issue.

Currently the integrations connection loss handle mechanism is rudimentary at best.
This means that in case of some (your example the battery / grid meter) slaves not reporting new values the integration should do a partial update. (of only the data that was able to be retrieved)
In case it loses the connection it will keep trying to retrieve new data and when it does the integration will publish the new data.
The availability of the entities and warning / error logging in case of failed connection/updates etc are part of the currently planned improvements. (As can be seen in the readme of this repository).

Therefore, this issue will be tackled when the connection handling gets improved (currently planned for the end of december / begin of january)

braamvh, 2022-12-12:
Thanks for the quick response. In the interim I am going to go back to my old modbus config until January or so.  If I could run my own modbus sensors and this integration, I would have just added the ones that gets incorrectly detected, but right now from the README that won't work.

I really like this integration, so I hope this isn't too difficult to take forward. It adds quite a bit of extra data that's timeconsuming to add in YAML by hand, so with this I have many more sensors than I had before.

sfstar, 2022-12-12:
Simplifying the yaml modbus configuration for users is the exact reason this integration was born ;).
I thought it to be better to put effort into an integration that simplefies the victron modbus setup in home assistant.
Rather than spending the same (or probably higher) amount of time finding available registers, writing yaml config and testing whether the yaml config was valid.

Having said that improving upon the connection resiliency isn't a high priority for me at the moment (I have the gx device connected via a lan cable to the switch on which homeassistant is connected).
Just yesterday I completed my physical system and I still need some time to build additional battery capacity / use the output of the integration in my automations.
Once that is done I will return to improving the connection resilience (hence the end of december / january timeline).

However, I'm always open to receiving a PR to add / improve upon the integration ;).

braamvh, 2022-12-13:
Heh, I am a network engineer, not a developer, my coding skills are fairly basic :) But who knows, this may fall in my capability.

About my setup that may explain some of the things I see:

- My CCGX is LAN-connected, so network connectivity is unlikely to be the issue. I probe it frequently on a network level, and the only times it misbehaves is when it auto-updates.
- I have a CCGX, not a Cerbo GX. I am not too sure if that is the cause of the issue, as the hardware in the CCGX is not as powerful. I am intending to upgrade to a Cerbo sometime next year.
- We have a lot of power outages in South Africa (hence the invterter in the first place). When this happens, the ET112 falls away and becomes unreadable. It's still detected by the integration, though, but with all values listed as "Unavailable". When it recovers, I have seen it register the current value, then never update again after that.
- The VE.Bus device never gives me issues, the entities update relatively quickly, so in general polling works.
- I use the default polling from the Modbus module, which according to the docs is 10s. Not sure if that makes a difference.

The initial battery issue never resurfaced and may be a once-off thing. What I did see, however, is that since the system device and the first Ve.CAN device have the same unit ID, the integration seems to get confused between them. Sometimes unit 0 shows up as system and unit 100 as solarcharger, other times both show up as solarcharger.  In that case the solar charger entities go unavailable as well, and I have not found a way to restore them, other than re-installing the integration. This happens most often when I resetart HA for some reason, mostly for updates.

Raini20, 2023-01-04:
I think I have the same issue, some entities become Unknown.
[image]
[image]

When the integration rescans it discovers the _100 device as _0. I can fix it with restarting and it keeps polling data for a few hours but randomly stops. I just read that restarting breaks it, but in my case restarting fixes it

braamvh, 2023-02-04:
I am going to close this, as I am not sure the integration can fix it. I have replaced my CCGX with a Cerbo and my lead-acid bank with a Li-Ion bank. I'll monitor and see if I need to open a new issue. The VE.Can issue is the same as #21

closed 2023-02-04 by braamvh