sfstar/hass-victron #10: Battery Sensor to use for Energy Dashboard

mr-birdie, 2022-12-14:
Good Day,

Thank you for this awesome integration. Loving it.

I am having a issue where I think not all my battery sensors are available for the energy dashboard.
These are the only ones showing, but they are reporting zero.

[image]

Is there something that I am missing? 

Also, can you maybe give the sensors that you use for your dashboard? All these sensors are quire confusing to me and I am not 100% sure which ones represent what.

Any help would be much appreciated

sfstar, 2022-12-14:
Thank you for providing the additional information when opening this issue.
Currently I'm using the following entities to track my batteries usage:
[image]

Whereby the acin2 to inverter doesn't see any current (my multiplus doesn't have an acin2).
Otherwise I couldn't add the feedin and the self consumption (Since HA doesn't allow for multiple in / out trackers in the same battery system).

Do you have an system with an can-bms battery? (Still waiting on shipping for my can-bms battery and currently using an voltage based highly inaccurate battery monitoring through the multiplusses battery monitoring).

Could you also provide me with a screenshot of your GX device available modbus devices? (under "menu" -> "settings" -> "services" -> "modbus tcp" -> "available services".
This will help in determining what entites should work best for battery tracking in your setup.

mr-birdie, 2022-12-14:
Thanks for the prompt response:

Below the modbus devices.
[image]

[image]

System Overview:

[image]

[image]

Grid is currently of (due to loadshedding, perks of living in South Africa)

I assume AC-in means from Grid and AC-Out is the AC Loads of my house?

sfstar, 2022-12-20:
Thank you for the additional information.
Yes the AC-in is `GRID/GENERATOR`
AC-OUT is `LOADS`

I just checked the victron modbus spec for any KWH based registers that you could use in your dashboard.
However, it seems that the battery unit doesn't expose KWH counters.
Currently (this might change if victron releases V3 of the gateway and the new modbus spec definition) your best option is to replicate my configuration with the vebus entities.

CTRDevelopments, 2023-01-08:
Hi, I hand coded all of this (there was no repository so great work). You will need to integrate the energy stats for the energy page

Something like

`  #total energy to battery
  - platform: integration
    source: sensor.battery_input_energy
    name: total_battery_input_energy
    unit_prefix: k
    method: left
    #or right or trapezoidal for methods
    round: 3`

mtucker502, 2023-01-31:
> Thank you for the additional information. Yes the AC-in is `GRID/GENERATOR` AC-OUT is `LOADS`
>
> I just checked the victron modbus spec for any KWH based registers that you could use in your [...]

Does this include Lynx 1000A Shunt?
[image]

[image]

It seems like the entities are present but have 0 values:
[image]

Or should I be using a different entity for tracking battery consumption?

mtucker502, 2023-03-27:
> Thank you for providing the additional information when opening this issue. Currently I'm using the following entities to track my batteries usage: [image]
>
> Whereby the acin2 to inverter doesn't see [...]

Do these vebus sensors account for MPPT charge controllers and the power going straight into the batteries that doesn't pass through the inverters?

mtucker502, 2023-04-25:
I have a BMS CAN battery (EG4 Communications Hub) and the battery charged/discharged values are not incrementing:

[image]