sfstar/hass-victron #11: Retrying setup: division by zero

Raini20, 2022-12-15:
Hello, I tried to use your integration over HACS with my newly installed Victron system but failed at the first step.

I used the IP address which is statet on the router page:
[image]

The port number is untouched:
[image]

It tells me Success! but your integration shows an error:
[image]

The log tells me nothing I understand:
[home-assistant_victron_2022-12-15T21-38-48.144Z.log](https://github.com/sfstar/hass-victron/files/10240839/home-assistant_victron_2022-12-15T21-38-48.144Z.log)

The Modbus TCP services page on my Cerbo GX tells me following ids:
[image]
[image]

The last Modbus/TCP error occurred at the time of configuration of the integration:
[image]

The Firmware of the Cerbo GX is v2.92
The Firmware of the MultiPlus-II 48/5000/70-50 is 497
The Firmware of the SmartSolar MPPT VE.Can 250/100 rev2 is v3.10

I don't know if I missed something important, but I reread the github page two times and found nothing.

sfstar, 2022-12-20:
Hello and thank you for opening this detailed issue.
Especially the logs were really nice to have.

Based on the logs I suspect that you might be using an older version of the integration.
Since this error might occur if a value is divided by a scale of 0.
This behaviour was changed in one of the latest releases because it turned out the spec has no registers with a scale of 0.
If you are running into this issue with the latest integration release please let me know.
In that case I will have to make a debug beta release so I can better troubleshoot your issue.

Thanks in advance

Raini20, 2022-12-21:
Hello I think I have the current version installed (v0.0.5):
[image]

[image]

I just clicked redownload and restarted, but it still gives me the same error

sfstar, 2022-12-23:
I've just created the v0.0.6.beta-1 release to troubleshoot your issue.
Please install it, restart HA, remove the integration, add the integration with debug logging enabled and submit the logs.
Thanks in advance

Raini20, 2022-12-23:
That's the log with the new version:
[home-assistant_victron_2022-12-23T12-24-47.730Z.log](https://github.com/sfstar/hass-victron/files/10294987/home-assistant_victron_2022-12-23T12-24-47.730Z.log)

Hope I did it right with the log

sfstar, 2022-12-23:
Thank you.
It seems that the version was correctly installed (the error is triggered on the same line of code that in the beta has a different line number).
However, it seems that debug logging wasn't active.
Could you enable it via the hamburger menu option -> "Enable debug logging":
[image]

And resubmit the logs again with the debug logging active?

Raini20, 2022-12-23:
Puh okay, I tried it again:
https://user-images.githubusercontent.com/68600283/209348030-c7cbe277-0c5d-45d8-bc17-e89828e15052.mp4

That's the log from the actions in the video, but I think it's not different at all:
[home-assistant_victron_2022-12-23T13-51-15.079Z.log](https://github.com/sfstar/hass-victron/files/10295491/home-assistant_victron_2022-12-23T13-51-15.079Z.log)

I'm not sure how else I should activate the debug logging

sfstar, 2022-12-23:
Thanks for the video (makes it easier to follow what you did).
Seems to me that you did everything correctly.
It should have generated lines like this:
```
2022-12-23 11:59:28.454 DEBUG (MainThread) [custom_components.victron.coordinator] decoding uint32 for 228.vebus_invertertoacin2
2022-12-23 11:59:28.454 DEBUG (MainThread) [custom_components.victron.coordinator] decoded value 0 with scale of 100 and for unit kWh
2022-12-23 11:59:28.454 DEBUG (MainThread) [custom_components.victron.coordinator] decoding uint32 for 228.vebus_invertertoacout
2022-12-23 11:59:28.454 DEBUG (MainThread) [custom_components.victron.coordinator] decoded value 213 with scale of 100 and for unit kWh
2022-12-23 11:59:28.454 DEBUG (MainThread) [custom_components.victron.coordinator] decoding uint32 for 228.vebus_outtoinverter
2022-12-23 11:59:28.454 DEBUG (MainThread) [custom_components.victron.coordinator] decoded value 0 with scale of 100 and for unit kWh
2022-12-23 11:59:28.454 DEBUG (MainThread) [custom_components.victron.coordinator] Finished fetching victron data in 0.046 seconds (success: True)
```

One more thing you could try is adding the following to your configuration.yaml:
```

logger:
  default: info
  logs:
    custom_components.victron: debug
```
After which you only need to navigate to `developer tools` and use the restart button there:
[image]

This will permanently enable debug logging via yaml configuration (whether the integration is removed and readded or just stays active - debugging will remain enabled for this integration)

After the restart via developer tools is done you need to wait a couple of seconds to allow for some logs to be generated.
After which you can use this button to navigate to your instance logs: [image]

And use the download logs button to get all system logs (or use load full logs to pick the relevant lines).

Raini20, 2022-12-23:
My updated configuration.yaml:
[image]

I just removed the integration and deleted it in HACS.
Now I restarted and added it back (v0.0.6.beta-1) [17:47:45]

Now I have plenty of DEBUG messages, I think this is what you were looking for:
[home-assistant_2022-12-23T16-50-35.344Z.log](https://github.com/sfstar/hass-victron/files/10296296/home-assistant_2022-12-23T16-50-35.344Z.log)

I didn't shorten it or something, it's the full log

sfstar, 2022-12-23:
Thank you for providing the logs.
This is exactly what is was looking for:
```
2022-12-23 17:49:20.064 DEBUG (MainThread) [custom_components.victron.coordinator] decoding uint16 for 100.generator_runningbyconditioncode
2022-12-23 17:49:20.064 DEBUG (MainThread) [custom_components.victron.coordinator] decoded value 0 with scale of 1 and for unit 
2022-12-23 17:49:20.064 DEBUG (MainThread) [custom_components.victron.coordinator] decoding uint16 for 100.generator_runtime
2022-12-23 17:49:20.064 DEBUG (MainThread) [custom_components.victron.coordinator] decoded value 0 with scale of 0 and for unit s
2022-12-23 17:49:20.064 ERROR (MainThread) [custom_components.victron.coordinator] Unexpected error fetching victron data: division by zero
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 225, in _async_refresh
    self.data = await self._async_update_data()
  File "/config/custom_components/victron/coordinator.py", line 78, in _async_update_data
    parsed_data = OrderedDict(list(parsed_data.items()) + list(self.parse_register_data(data, register_info_dict[name], unit).items()))
  File "/config/custom_components/victron/coordinator.py", line 98, in parse_register_data
    decoded_data[full_key] = self.decode_scaling(decoder.decode_16bit_uint(), value.scale, value.unit)
  File "/config/custom_components/victron/coordinator.py", line 122, in decode_scaling
    return number / scale
ZeroDivisionError: division by zero
```

So the issue is the generator runtime register.
You must be the first user trying the integration with a generator included in your setup 😉.
The seconds should be returned as is and due to an earlier interpretation error (that was already partially fixed see: #7 for more info) the scale value was still set to 0 instead of 1 for generator_runtime and generator_runtime_2 registers.

This causes the behaviour seen by you since the integration (as a result of patching #7) no longer has a mechanism to handle a scale of 0.
Since it turned out that the modbus spec I originally based the integration on incorrectly defined some scales as 0.
While in actuality there are (as of today) no registers with a scale of 0.

I'm making a beta patch and will notify you when the fix can be tested.

sfstar, 2022-12-23:
Please install `v0.0.6.beta-2` and restart homeassistant after (via the dev tools).
If the integration doesn't suddenly start working or doesn't show all your devices you could remove and add the integration again.
After which the setup should now complete without the division by zero issue.
Please let me know if the beta-2 fixes your issue so that i can make a proper release to fix this issue 😉

Raini20, 2022-12-23:
It worked!!
[image]

My system is currently in standby so I can't verify the values except the battery voltage, but the setup process finished without problems and it shows everything. Thank you

sfstar, 2022-12-23:
Great, I will promote the changes to a full release (without the additional debugging code).
Feel free to close this issue after you validated the values when the system is not in standby (or report back if the values are incorrect).

sfstar, 2022-12-23:
v0.0.6 has just been released and the beta candidates have been purged

Raini20, 2022-12-23:
v0.0.6 works and the values seem to be correct, thank you again

closed 2022-12-23 by Raini20