sfstar/hass-victron #13: pvinverter pvtotal

Aikawa24, 2022-12-22:
hi, ive a problem with the measurement of pvinverter pvtotal it shows 130KW but it should show 0,13KW

sensor.victron_pvinverter_power_total_20

the sensors 
sensor.victron_pvinverter_l1_power_20
sensor.victron_pvinverter_l2_power_20
sensor.victron_pvinverter_l3_power_20

shows correct in W

can you fix this please?

thank you for your work :-)

sfstar, 2022-12-22:
Hello and thank you for opening this issue.
I've also got a pvinverter which i just added to the vrm environment and I'm seeing the same behaviour.
However, it seems that this is a victron GX issue for the specific register.

Since the register definition states:
[image]

Which in the home assistant core integration modbus yaml would translate into (with off course your own gx device ip instead off <ip>):
```
 - name: victron
   type: tcp
   host: <ip>
   port: 502
   retry_on_empty: true
   retries: 10
   close_comm_on_error: true
   sensors:
      - name: victron_pvinverter_L1_power
        unit_of_measurement: "W"
        address: 1029
        slave: 30
        device_class: power
        state_class: measurement
        data_type: uint16   
      - name: victron_pvinverter_power_total
        unit_of_measurement: "kW"
        address: 1052
        slave: 30
        device_class: power
        state_class: measurement
        data_type: int32
```

Which also returns the incorrect scaled value for the power_total register.
So either there is a error in the documentation of victron or there is a bug in the modbus tcp server code of the GX device.

I'm affraid I can't solve this issue without starting to assume behaviour of the GX device.
Although this could solve our problem, assuming the actual intended data format outside of the official spec could cause issues down the line.
For example if victron patches this bug (in a newer version) then the integration would be reporting incorrect values for that newer version.

Therefore, I'm not going to change (/allow any changes) that would force the integration to diverge from the victron modbus TCP spec.
If victron were to update the spec then I'm willing to patch the integration to match it.
For now the best thing to do is to make victron aware of this bug so that it might get resolved.

sfstar, 2022-12-22:
Linking victron community bug report issue: https://community.victronenergy.com/questions/176336/bug-modbus-pvinverter-device-reports-incorrect-val.html

Aikawa24, 2022-12-22:
Ive a solaredge se16k inverter maybe the bug comes from the modbus connection between the inverter to the gx device and then goes the way to Home assistant

Aikawa24, 2022-12-22:
[image]

Aikawa24, 2022-12-22:
[image]

sfstar, 2022-12-22:
@Aikawa24 good suggestion.
However, I'm seeing the same behaviour in my setup where the pvinverter is a et112 configured as a pvinverter on output.
Therefore, I'm even more sure that it is something inside of the GX modbus implementation.
Otherwise both the et112 (which is a serial modbus device) reported values and the solaredge reported values should be incorrect.
Which seems unlikely to me.

This is however great information in further narrowing down this issue (Thanks @Aikawa24)

OlePZ9, 2022-12-23:
Hi, I can confirm with Fronius Inverters . I entered "W" as unit to get the correct result

[image]
As a control, I adden a Sum Sesor for the three phases
[image]
[image]

OlePZ9, 2022-12-23:
[image]

OlePZ9, 2022-12-24:
only this to be changed?
[image]

OlePZ9, 2023-01-03:
[image]

sfstar, 2024-03-15:
Closing as Issue should be resolved in release v0.1.7
Feel free to re-open if issue re-occurs at your end.