sfstar/hass-victron #14: How does your integration handle an existing Modbus.yaml configuration?

OlePZ9, 2022-12-22:
Hello, I am very happy that I found your new integration! I know how much hard work it is to get all this stuff in a yaml file..
Due to the fear of losing this work, I'd like to get an idea what will happen with my existing work after having performed the installation ? Thanks so much!

sfstar, 2022-12-23:
Hello @OlePZ9,

The following is a snippet of the readme:
```
## Warning
This integration uses pymodbus 3.0.2 or higher.
As of november 2022 the built-in home assistant modbus integration runs on a version < 3.0.0
If you install this integration the built-in modbus integration will stop to work due to breaking changes between 2.x.x and 3.0.0
```

This means currently you cannot use this integration and the yaml modbus config at the same time.
This is expected to be solved around january (the core modbus integration is planned to be updated after pymodbus 3.1.0 is released, which is planned for this month).
You can try out this integration without losing you configuration by commenting uit the modbus yaml definition (by prepending every modbus yaml line with a #).
Then you can restart and install this integration.

Switching back to your old config would be the same in reverse.
Removing this integration.
uncommenting modbus yaml.
Restarting home assistant.

Another solution would be to setup https://github.com/custom-components/remote_homeassistant
And run each integration in its own homeassistant instance.

sfstar, 2022-12-23:
I will keep this issue open, until the core integration has migrated, since this question seems to come up often in several different issues

OlePZ9, 2022-12-23:
Thank you very much! I appreciate your work!

flyrmyr, 2023-02-01:
@sfstar I just dropped my custom Modbus yaml in favor of your integration, and I gotta say thanks so much for your effort, it's awesome! Thought about building something like this for a while.

I'm still rewiring my values back up, but your work slots in perfectly with the Lovelace element I've been building. It brings the live power flow concept from the Cerbo into a design similar to the HA Energy dashboard.

[image]

FuzzyB3ar, 2023-02-08:
So glad I found this integration!

@flyrmyr would you mind sharing your yaml for this card?

flyrmyr, 2023-02-08:
@FuzzyB3ar The card info can be found here:
https://github.com/flyrmyr/system-flow-card

Although the ReadMe isn't fully converted from the work, and the card is still in late beta ish stage., you can follow the progress here:
https://github.com/flyrmyr/system-flow-card/issues/1

closed 2023-05-05 by sfstar