sfstar/hass-victron #15: [Feedback] 3 phase ESS

nevesenin, 2022-12-23:
Hi!

First of all, thank you. I've been waiting for this integration.

Since the README stated that it has not been tested with a 3 phase system, I'm just guessing that this is related to a 3 Phase system.

So, setting up the integration worked flawlessly. By now I only did this. No further configuration. Also, there are no errors or any issues.

I recognized two minor things that I find confusing and want you to know. On the devices overview the battery SOC values are taken from the wrong sensor.
[image]
For the battery device it seems to be the value from `victron_battery_midvoltagedeviation` instead of `victron_battery_soc`:
[image]
And for settings its `victron_settings_ess_batterylife_soclimit` but should be `victron_system_battery_soc`:
[image]

My current setup:
Home Assistant 2022.12.8
Supervisor 2022.11.2
Operating System 9.4
HACS 1.29.0

Victron CerboGX with v2.29 Firmware
3x Pylontec US3000C Batteries

sfstar, 2022-12-23:
Thank you for opening this issue and taking the time to provide feedback.

Currently I'm waiting on my utility provider to install a three phase net connection in my home.
Therefore, I have my 3x multiplus 48/5000 setup in parallel for now.
This means that I currently am unable to validate the values provided by the three phase specific entities.
Although they should work the same as my L1 entities, there might always be some small typo etc that causes L2 and L3 entities to not report the correct values.

I'm aware of the incorrect value being taken by the devices to display SOC.
It seems that the device takes the first SOC entity in alphabetical order.
Will need to look into whether I can specify the SOC entity to use for a homeassistant device definition.

If you find any other issues or a missing something with your setup please let me know.
Having this feedback is greatly appreciated 😉

nevesenin, 2022-12-24:
> If you find any other issues or a missing something with your setup please let me know.

Thanks, I'm going to do so ;).

I just updated to v0.0.6 and realized another minor thing, which currently only seems to be an UI issue. After updating and restarting HA the battery device also is named `settings`.
[image]
I tried to figure out why that is and maybe provide a PR, but I quickly realized that I need more time to get used to HA ecosystem and Python in general. But I'd like to share my thoughts on that, even if there is still a lot of guessing involved.
I assume that this specific issue is due to the device registration through entities, which adds a lot of complexity in the workflow if there are a lot of entities and one want's to be explicit about the device. In that regards and as a side note, I wonder if there shouldn't also be a `system` device (definitely a guess) in the device list.
I would like to hear your opinion on explicitly register devices before entities. That would also offer a way to have e.g. a ESS specific device that combines entities from `vebus` and `settings` by mapping entities (addresses) to devices. Could also be an advantage when upgrading to GX firmware v3.

closed 2025-03-05 by github-actions[bot]