sfstar/hass-victron #16: installation issue

OlePZ9, 2022-12-23:
I am trying to install it on a backup system. the original Modbus integration had been disabled. I was able to install, but unable to configure the integration. I am using a 3 phase MPII 5000/48, cerbo GC, Pylontech LiFePO4 battery, 2x Fronius inverter (Symo 3 Phase), 1x MPPT 450/100, 1x MPPT 250/60. The Yaml integration had been disabled. Any idea?
[image]

OlePZ9, 2022-12-23:
[image]

Raini20, 2022-12-23:
Is your Modbus/TCP under "Settings -> Services" enabled and further under " -> Modbus/TCP -> Available Services" are your devices in the list?

OlePZ9, 2022-12-23:
In another instance of HA my modbus connection is workin flawlessly. This is a test of the integration after removing all Modbus stuff from yaml

OlePZ9, 2022-12-23:
Maybe I need to shutdown the other instance...

OlePZ9, 2022-12-24:
I think, I got it: enter http:// before the IP address :-)

closed 2022-12-24 by OlePZ9