sfstar/hass-victron #17: First Feedback and idea for improvement

OlePZ9, 2022-12-24:
3 Phase is working well. My first proposal for improvement wuld be to add a unique ID (could be the same as entity, this would help to add them into dashboards. Additionally, the ability to add all (or more than one a a time) entity to a dashboard would help
[image]

Add to dashboard:
[image]

sfstar, 2022-12-31:
Hello and thank you for opening this issue.
I'm not sure why the unique id is not being set for you.
In my own production setup (running the latest released version) every entity (including my grid energy forward total entity) has a unique id.
[image]

Is the unique id not being set (at your end) occuring for every entity provided by this integration?

OlePZ9, 2023-01-03:
I reinstalled the integration and now it works from this point. This time the IP Address (without http:// before) gave me success. But now, I have another Issue. Will open a new one and close this one, thanks!

closed 2023-01-03 by OlePZ9