sfstar/hass-victron #18: Battery Pylontech - Charge and Discharge does not shows data @ Energy Dashboard

VSLRA, 2022-12-27:
PYlontech Battery (com.victronenergy.battery) Unit ID:225 does not show any data. In the device data overview I see 0,0 kWh for both sensors.
- sensor.victron_battery_history_dischargedenergy
- sensor.victron_battery_history_chargedenergy

Some other entities related to battery works (See att.)

Therefore integration to Energy Dashboard does not work properly.

sfstar, 2022-12-31:
Hello and thank you for opening this issue.
I see that you have found that can BMS'es unfortunately don't populate the charge and discharge registers.
While they still are available to be queried by this integration.

As to you issue with setting the minimum soc.
I just tested setting the minimum soc from my production instance and everything worked as intended.
There was only a small "bug" that I set the value to 27% and that the vrm console showed the value to be set as 25%.
Most likely victron only displays or allows setting this value to a multiple of 5.

In case you can't set the batterylife minimumsoc entity make sure that you enable write support in the integrations configuration.

CTRDevelopments, 2023-01-08:
I dont think it will accept any value that is not a multiple of 5. I dont think it rounds, it just errors the command

sfstar, 2023-03-06:
This issue should now be adressed in the latest release containing the fix made by @ChristophPech.
I will close this issue if there are no further reports in the coming weeks.

closed 2023-05-05 by sfstar