sfstar/hass-victron #19: Charge control according to market prices

Beny44, 2022-12-31:
Perfect work.
I have a question about controlling the charging according to the price of electricity on the market.
I have an integration into our market operator where I can see the current price and the price for the day.
I would like according to some key to enable charging in the low price window.
Would it be possible to work this into the integration somehow?

Thank you and have a great New year. 

Translated with www.DeepL.com/Translator (free version)

sfstar, 2023-01-03:
Hello and thank you.
The integration provides all available victron modbus registers.
If you enable write support in the integrations config all writeable entities should become available as such.
These entities can be used to modify charge behaviour (for example I use the number.victron_settings_ess_acpowersetpoint entity to determine when to charge / consume from grid, when to compensate all loads in the house and when to sell back energy to the grid.

As to how you can instruct this behaviour:
Homeassistant comes with multiple methods (for example the node-red addon or the built-in automations setup) to run specific actions based on specified triggers and conditions.

This integration is an atomic building block in smartifying your energy consumption.
It's core and only function is to simplify and expedite the setup of the modbus integrations registers with homeassistant.
Other integrations (Like the [entsoe-e](https://github.com/JaccoR/hass-entso-e) integration) provide other building blocks that you can use in your homeassistant instance.

So in short you can use the existing entities this integration provides to achieve what you want to do.
But combining the low power window and changing registers in this integration will have to be done via setting up your own automations in your preferred tool (node-red/built-in automations).

sfstar, 2023-05-05:
Victron is testing an feature in a beta release that allows you to optimize charging based on dynamic pricing and forecasting between self-consumption and production.
Furthermore, the feature request is outside the scope of this integration and can be implemented with homeassistant automations and the entso-e integration.
Therefore closing this issue.

closed 2023-05-05 by sfstar