sfstar/hass-victron #20: Battery sensors shows mostly zero values

kucerazdenek, 2023-01-02:
Hello,

first let me announce that I am newbie in HA as such. During last week I have been able to integrate Stiebel Eltron heat pump and jumped to PV. My configuration is 3-phase, Fronius Symo 8.2.3 (currently without datalogger card, Fronius currently distributes only Light version of their inverters, thus I am not able to read data for PV production via modbus till end of Q1/2023 ), 3x Victron Multiplus II/3000 + 2x BMZ ESS 9.0 + Cerbo GX.

After victron integration installation, I entered IP of Cerbo, and system get scanned correctly. Now I see 4 devices, all sensors seems to be reading correctly except "battery device". 

[image]

Alarm sensors on battery device reporting OK, few status sensors also (temperature, soc, settings etc,) but on the rest of sensors, I see zero values.
Do you have some advise what I am doing wrong?

[image]
[image]
[image]
[image]
[image]

Thanks a lot.
Zdenek

sfstar, 2023-01-02:
Hello and thank you for opening this issue.
If you are using a CAN.bus BMS than the following applies for the entities that are 0: `Not supported by CAN.bus BMS batteries.`.
These entities are still created by the integration since the registers are available when a CAN.bus BMS is used.
However, they aren't populated with data by victron in this mode.

If using an other type of mode to connect your battery please let me know so we can troubleshoot what is wrong.

kucerazdenek, 2023-01-02:
Thanks for you quick reply! I just checked the connection (PV was built by supplier company, not by me) and you are right, on BMZ batteries, there is port marked as CAN (batteries are connected as cascade through it) and from first battery this port is connected via STP cable to Cerbo GX. Your explanation makes sense, so my question is: should I read values for battery from vebus device? Into energy dashboard finally I configured "vebus outtoinverter" as energy going to battery and "vebus invertertoacin1" as energy pulled from battery. Those values are closest (but not exact) to data I see in VRM (but I am not sure if I use correct sensors in HA)
Basically I raised this topic, because I was not sure what sensors have to be used for energy view/battery and also wanted to create own dashboard in HA and assumed that data which I am able see in VRM, should be also available as sensors in HA, because VRM must use very same data provided by Cerbo GX. i.e. in VRM I am able to see current level of charging/discharging as real-time value, but not sure what sensor is it in HA..... Also I am interested in historical statistic split of AC load (heat pump + Air condition) and critical/backed up load (rest of home), which cannot be done in VRM even those values are provided in realtime VRM dashboard.
Example here (its night here and battery in min SoC, thus there is no energy traffic from/to battery now)

[image]

sfstar, 2023-01-03:
The Vebus entities seem to be the only registers available that provide charge/discharge information when using CAN bus connected batteries.
Currently I'm still waiting on a can bus BMS that I will integrate with my setup.
So until then I'm unable to find out if there are any other/better options to use.

My setup is voltage regulated (with an BMS that doesn't communicate with victron).
Therefore, I'm currently also using the Vebus registers in my energy dashboard.

Based on the modbus spec it seems that if you are using  can bus BMS you are out of luck regarding the most helpfull / precise data regarding charge / discharge totals.
However, I'm still hoping that when I hookup my can bus BMS it will populate the vebus registers with the information it gets from the can bus bms instead of "guesstimating".

As to values differing between the VRM console and the modbus integration:
I've seen some registers rounding returned values to the nearest fully divisible by 10 number.
So this might explain why some values (that aren't guesstimated) will differ slightly from what you are seeing in VRM.

Frankly, while modbus is documented as the preferred third party integration mechanism, I find it really lacking precision and the range of information that is queryable (when compared to the data inside VRM itself)

arthur2017-ui, 2023-02-10:
Hello, I am using your Victron GX modbusTCP integration. Good job ! it works fine. Thank you very much.
Except in "battery" device for the 2 entities History chargedenergy and History dischargedenergy that remain 0.0 kWh.  So I cannot use them directly in the HA energy management configurator and I don't see any flow on the battery.
From your integration, I got 6 devices and 312 entities
[image]
[image]

But I have good news because  I get in the "settings" device the power (W) = "system battery power" which is transfered to battery (positive number when "system battery state" = CHARGING) or from battery (negative value when "system battery state" = DISCHARGING) under the entity "system battery power". The numbers of "system battery power" in W are exactly the same numbers that I have on my VRM dashboard.
[image]

So I believe that it would be possible with Riemann to convert from "system battery power" the Power to Energy... I tried in a simple way (just with the value of system battery power), but it doesn't work. 
 
I am not enough envolved for using code in config.yaml file to use properly the sensor.victron_system_battery_power and Riemann, i.e. to calculate Battery_Charge_Energy and Battery_Discharge_Energy in order to suit directly the 2 excpected values in HA Energy Dashboard.

Could you please let me know either when you expect to include the 2 missing entities "History charged/discharged energy" in a next upgrade of your integration ; or the yaml code to determine the 2 entities for using them directly in HA Energy manager under Energy In and Energy Out the battery.

Thank you a lot in advance.

René

kucerazdenek, 2023-02-16:
> Hello, I am using your Victron GX modbusTCP integration. Good job ! it works fine. Thank you very much. Except in "battery" device for the 2 entities History chargedenergy and History dischargedenergy that remain 0.0 kWh. So I cannot use them 

Hi René, I tried this and works fine.
**configuration.yaml:**

```
      - name:  "VRM - Battery to House"
        unique_id: vrm_battery_to_house
        state_class: measurement
        device_class: power
        unit_of_measurement: 'W'
        state:  >
          {% if states('sensor.victron_system_battery_power') | float(0) < 0 %}
            {{ states('sensor.victron_system_battery_power') | float(0)*-1 }}
          {% else %}
            0
          {% endif %}          
          
      - name:  "VRM - PV/Grid to Battery"
        unique_id: vrm_pv_grid_to_battery
        state_class: measurement
        device_class: power
        unit_of_measurement: 'W'
        state:  >
          {% if states('sensor.victron_system_battery_power') | float(0) > 0 %}
            {{ states('sensor.victron_system_battery_power') | float(0)* 1 }}
          {% else %}
            0
          {% endif %} 

```

**sensors.yaml**

```
#VRM stats
  - platform: integration
    source: sensor.vrm_battery_to_house
    unique_id: vrm_battery_energy_to_house
    method: left
    unit_prefix: k
    name: VRM - Battery Energy to House

  - platform: integration
    source: sensor.vrm_pv_grid_to_battery
    unique_id: vrm_pv_grid_energy_to_battery
    method: left
    unit_prefix: k
    name: VRM - PV/Grid Energy to Battery
```

arthur2017-ui, 2023-02-17:
Thank you for your reactivity. 
Two days ago I finaly got the energies input and output of the battery with the following code that looks like yours. 
Is it exactly the same or would I get a better result with yours ? Because with my code, HA gives me a higher result than VRM: +1.1 kWh for output and 0.4 kWh for input per day.
What is your opinion ?

sensor:
 #total battery input energy
  - platform: integration
    source: sensor.battery_power_input
    name: battery_input_energy
    unit_prefix: k
    method: left
    round: 3

    #total battery output energy
  - platform: integration
    source: sensor.battery_power_output
    name: battery_output_energy
    unit_prefix: k
    method: left
    round: 3

template:
    - sensor:
        name: "Battery power input"
        device_class: power
        unit_of_measurement: "W"
        state: >
          {% if states('sensor.victron_system_battery_power')|float >= 0 %}
            {{ states('sensor.victron_system_battery_power') }}
          {% else %}
            0
          {% endif %}
    
    - sensor:
        name: "Battery power output"
        device_class: power
        unit_of_measurement: "W"
        state: >
          {% if states('sensor.victron_system_battery_power')|float < 0 %}
            {{ -1 * states('sensor.victron_system_battery_power')|float }}
          {% else %}
            0
          {% endif %}

[image]

kucerazdenek, 2023-02-17:
This is exactly the same like I did. Only difference is that you have specified round, but it is 3 in default as well.
I am far away to judge which measurement is more precise, if VRM or this one.
Anyway, it is much more closer to VRM than my previous setup, where I used accin and accout sensonsors provided by Victron VeBus

closed 2025-03-05 by github-actions[bot]