sfstar/hass-victron #21: entities diappear

OlePZ9, 2023-01-03:
After having installed the integration successfully, the integration found 509 entities, both solarchargers included. After a while (reboot?) some of the entities disappeared.
[image]
[image]
[image]
[image]

What happened here?

OlePZ9, 2023-01-03:
[image]

Since adding the integration, this all is not working anymore...

OlePZ9, 2023-01-03:
[image]

sfstar, 2023-01-03:
What are the errors you see when you click the logs link that you can see in your screenshot?
Are there any errors on that page that were generated by this integration?

OlePZ9, 2023-01-03:
Honestly speaking, I have no idea what the reason is. The same setup with my manual Modbus integration in another virtual machine clone works fine

OlePZ9, 2023-01-03:
[image]

OlePZ9, 2023-01-03:
[image]

OlePZ9, 2023-01-03:
[image]

OlePZ9, 2023-01-03:
Braamvr may had explained the reason in his issue about missing data from his grid meter

OlePZ9, 2023-01-03:
[image]

OlePZ9, 2023-01-03:
As you can see in the second picture, the integration also uses 0, in my Modbus setup I am using 100

OlePZ9, 2023-01-03:
https://www.victronenergy.com/live/ccgx:modbustcp_faq

OlePZ9, 2023-01-03:
[image]

sfstar, 2023-01-04:
Thank you,
This information is really helpfull in understanding what is going on at your ends.
I currently do not have a ve.can devices (planned for the near future) so in order to reproduce your issues I would need to check if I can reproduce it in demo mode.
Since this has a large impact on the functioning of my system I will need some time in order to find a suitable window for down-time.

OlePZ9, 2023-01-04:
Thank you for spending time on that! I really appreciate your work and altogether everything looks very promising! If you haben an idea where to change something, I could check, as my productive system runs idependently from the testing environment. Actually it would be too hard to change all names of my Victron entities and dashboards to get it working on my system. My yaml is about 2000 lines....

Raini20, 2023-01-21:
Hello I just made a little workaround for the same problem:

`automation reload_victron:
    alias: reload victron
    description: 'reload integration if entity unknown/unavailable'
    trigger:
      - platform: state
        entity_id: sensor.victron_solarcharger_yield_power_100
        to:
          - "unknown"
          - "unavailable"
    action:
        service: homeassistant.reload_config_entry
        data: {}
        target:
            entity_id: sensor.victron_solarcharger_yield_power_100
    mode: single`
    
    I wrote this in my configuration.yaml and if the trigger entity becomes unavailable/unknown it reloads the integration. For me it automates the reloading and I don't have any data holes anymore

askpatrickw, 2023-09-07:
Not sure if this new information or not ... 

I had been working all day with some GPS values from my Cerbo:  `states('sensor.victron_gps_latitude_0')`

After upgrading to 2023.9 all my `_0` were no longer active new `_100' entries existed.
`states('sensor.victron_gps_latitude_100')`

sfstar, 2024-05-19:
The issue has been shipped in the latest release (v0.3.0) closing issue as resolved.
Feel free to re-open if the issue persists with the v0.3.0 release

closed 2024-05-19 by sfstar