sfstar/hass-victron #22: Device class seems wrong for some sensors

nijel, 2023-01-05:
Some entities are having device class `voltage` and they have no unit. The device class seems wrong in this case, as there is definitely no unit for things like active input or overload alarm.

Sample log entries (there are way more of them):

```
Entity sensor.victron_vebus_activein_activeinput_227 (<class 'custom_components.victron.sensor.VictronSensor'>) is using native unit of measurement '' which is not a valid unit for the device class ('voltage') it is using; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.

Entity sensor.victron_vebus_alarm_l3_overload_227 (<class 'custom_components.victron.sensor.VictronSensor'>) is using native unit of measurement '' which is not a valid unit for the device class ('voltage') it is using; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
```

nijel, 2023-01-05:
The wrong class also makes Home Assistant try to graph the string entities like battery state instead of showing their string value in the history:

[image]

sfstar, 2023-02-03:
Thank you for reporting this issue.
Please see #25 for a explaination regarding the root cause and the commit linked in #42 for the fix to this issue.
Since posting in #25 i've come to notice a lot of the currently opened issues seem to be caused by this bug.
Therefore (in contrast to my explaination) I will be making the 0.0.8 patch release so that issues #25, #32, #42 and #43 can be closed

sfstar, 2023-03-06:
I will close this issue as resolved.
Since there were no new related issues opened (or posts made) in the past month.
Feel free to re-open if the issue persists

closed 2023-03-06 by sfstar

nijel, 2023-03-07:
Still seeing this with the current version (0.0.9), for example:

```
Sensor sensor.victron_vebus_alarm_l3_ripple_227 has device class None, state class None and unit thus indicating it has a numeric value; however, it has the non-numeric value: OK (<class 'str'>); Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
Sensor sensor.victron_vebus_alarm_phaserotation_227 has device class None, state class None and unit thus indicating it has a numeric value; however, it has the non-numeric value: OK (<class 'str'>); Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
Sensor sensor.victron_vebus_alarm_gridlost_227 has device class None, state class None and unit thus indicating it has a numeric value; however, it has the non-numeric value: OK (<class 'str'>); Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
```

Home Assistant still considers these as number when displaying history:

[image]

Such entities should have `None` as unit, not an empty string. https://github.com/sfstar/hass-victron/pull/58 does that.

PR #58 (2023-05-01): Fixed handling of non-numeric entities
The unit should be None in that case and not an empty string.

Fixes #22

* `unit=""` should be used for numerical entities without unit
* `unit=None` should be used for non-numerical entities