sfstar/hass-victron #25: System Charger Device ID 226 - 5 x Wrong Device Class entities

DaveAuld, 2023-01-18:
Device class set as Voltage - however these are actual general text states I believe, going by the data showing in Developer Tools   States, current values for me showing as;
Charger Alarm 226 = OK
High Voltage Alarm 226 = OK
Low Voltage Alarm = OK
Error Code 226 = NONE
MPP Operation Mode = OFF

Extract from the Log;
```

Logger: homeassistant.components.sensor
Source: components/sensor/__init__.py:980
Integration: Sensor (documentation, issues)
First occurred: 13:29:30 (86 occurrences)
Last logged: 13:29:30

Entity sensor.victron_solar_system_charger_alarm_226 (<class 'custom_components.victron.sensor.VictronSensor'>) is using native unit of measurement '' which is not a valid unit for the device class ('voltage') it is using; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.

Entity sensor.victron_solar_system_charger_alarm_lowvoltage_226 (<class 'custom_components.victron.sensor.VictronSensor'>) is using native unit of measurement '' which is not a valid unit for the device class ('voltage') it is using; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.

Entity sensor.victron_solar_system_charger_alarm_highvoltage_226 (<class 'custom_components.victron.sensor.VictronSensor'>) is using native unit of measurement '' which is not a valid unit for the device class ('voltage') it is using; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.

Entity sensor.victron_solar_system_charger_errorcode_226 (<class 'custom_components.victron.sensor.VictronSensor'>) is using native unit of measurement '' which is not a valid unit for the device class ('voltage') it is using; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.

Entity sensor.victron_solar_system_charger_mppoperationmode_226 (<class 'custom_components.victron.sensor.VictronSensor'>) is using native unit of measurement '' which is not a valid unit for the device class ('voltage') it is using; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
```

sfstar, 2023-02-03:
Thank you for reporting this issue.
It should be solved in the next patch release.
The conditional syntax for establishing the device class was incorrect.
instead of matching the entity unit to the word "voltage" it searched for "" in the word "voltage" (always returning true).
Therefore the default unit = "" for several status entities were set to "voltage" instead of none.
Also addresses #32, #42 and #43

sfstar, 2023-02-03:
v0.0.8 is now released with a fix for this issue.
Please install it to resolve this issue.
For now I will be closing this issue report in favor of #22 since it has the same root cause.
Feel free to re-open if the issue isn't resolved after updating and restarting.

closed 2023-02-03 by sfstar