sfstar/hass-victron #27: dcsource starter voltage Wrong value show 1.26 V instead of 12.6 V

Next9999, 2023-01-22:
dcsource starter voltage Wrong value show 1.26 V instead of 12.6 V

sfstar, 2023-02-01:
thank you for reporting this issue.
According to the victron gx modbus specification the scalefactor should be 100.
Which this integrations configuration reflects (see screenshots)
[image]

Although it seems that victron has either a definition or a specification bug for this to result in 1.26v instead of the 12.6v
Currently I'm still investigating whether I can get into contact with victron to get some clarification since this doesn't seem to be the first incorrect definition or even a buggy value being returned.

sfstar, 2024-05-19:
The issue has been shipped in the latest release (v0.3.0) closing issue as resolved.
Feel free to re-open if the issue persists with the v0.3.0 release

closed 2024-05-19 by sfstar