sfstar/hass-victron #28: GPS Altitude Wrong Unit of measure

Next9999, 2023-01-23:
GPS Altitude Wrong Unit of measure

m/s should be meter only

[image]

flyrmyr, 2023-02-01:
I've noticed the same with my manual setup in the past. The Cerbo is actually passing that as the unit value... Seems the error might be with VenusOS. Also, my value is just wildly off. Head scratcher.

sfstar, 2023-02-03:
Thank you for reporting.
It indeed is an incorrect defintion in the modbus specification sheet
[image]
Please see my comment in #36 as to why fixing this in the integration is currently on hold

askpatrickw, 2023-09-05:
Looks like Victron isn't engaging on the open issue, How about we test a fix?

Is the fix to change `"gps_altitude": RegisterInfo(2808, INT32, UnitOfSpeed.METERS_PER_SECOND, 10)` to "gps_altitude": RegisterInfo(2808, INT32, UnitOfSpeed.METERS, 10)

https://github.com/sfstar/hass-victron/blob/3b7fd201de6d7bc2daac1fd51b51ec46990660e3/custom_components/victron/const.py#L618C81-L618C81

askpatrickw, 2023-09-05:
BTW, the attributes.csv in the Victron repo says meters "m" now.

https://github.com/victronenergy/dbus_modbustcp/blob/master/attributes.csv#L7
`com.victronenergy.gps,/Altitude,d,m,2808,int32,10,R`

askpatrickw, 2023-09-05:
https://github.com/victronenergy/dbus_modbustcp/issues/31 is now closed and they mention they will update the docs accordingly.

sfstar, 2024-03-14:
Closing issue as fixed

closed 2024-03-14 by sfstar