sfstar/hass-victron #32: Entity sensor.victron_inverter_alarm_ripple_239 (<class 'custom_components.victron.sensor.VictronSensor'>) is using native unit of measurement '' which is not a valid unit for the device class ('voltage') it is using; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author. 9:38:37 PM – (WARNING) Sensor - message first occurred at 9:38:36 PM and shows up 105 times

Next9999, 2023-01-23:
ON a restart

Logger: homeassistant.components.sensor
Source: components/sensor/__init__.py:980
Integration: Sensor (documentation, issues)
First occurred: 9:38:36 PM (105 occurrences)
Last logged: 9:38:37 PM

Entity sensor.victron_inverter_alarm_ripple_239 (<class 'custom_components.victron.sensor.VictronSensor'>) is using native unit of measurement '' which is not a valid unit for the device class ('voltage') it is using; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
Entity sensor.victron_inverter_info_firmwareversion_239 (<class 'custom_components.victron.sensor.VictronSensor'>) is using native unit of measurement '' which is not a valid unit for the device class ('voltage') it is using; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
Entity sensor.victron_inverter_info_productid_239 (<class 'custom_components.victron.sensor.VictronSensor'>) is using native unit of measurement '' which is not a valid unit for the device class ('voltage') it is using; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
Entity sensor.victron_inverter_info_state_239 (<class 'custom_components.victron.sensor.VictronSensor'>) is using native unit of measurement '' which is not a valid unit for the device class ('voltage') it is using; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
Entity sensor.victron_inverter_alarm_lowsoc_239 (<class 'custom_components.victron.sensor.VictronSensor'>) is using native unit of measurement '' which is not a valid unit for the device class ('voltage') it is using; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.

joydashy, 2023-01-26:
You know you can edit the title before smashing enter, right?

sfstar, 2023-02-03:
Thank you for reporting this issue.
It has the same root cause as the bugs mentioned in #43 and #42 and will therefore also be fixed in the next patch release

sfstar, 2023-02-03:
v0.0.8 is now released with a fix for this issue.
Please install it to resolve this issue.
For now I will be closing this issue report in favor of #25 since it has the same root cause.
Feel free to re-open if the issue isn't resolved after updating and restarting.

closed 2023-02-03 by sfstar