sfstar/hass-victron #33: No entities?

JBsDk, 2023-01-23:
Hi,
I installed and configured the integration.
Then I enabled Modbus on the CerboGX.
The integration could be added in Home Assistant, but I don't see any entities... What can be the reason?
My CerboGX is connected to the Raspberry Pi 3B via a VEDirect to USB cable.

MasterXDave, 2023-01-26:
I just disabled „MQTT auf LAN (Klartext)“ and it worked. 
[image]

JBsDk, 2023-01-29:
@MasterXDave 
Danke für die Antwort,
das hat bei mir aber nicht funktioniert.
Was hast du sonst noch für Einstellungen vorgenommen?
Ich kriegs einfach nicht hin, weder mit der Integration noch manuell per Config-Definition ❌

sfstar, 2023-02-01:
Hello and thank you for opening this issue.
Please enable the modbus tcp service before configuring the integration and see if it then starts to work.
If not please enable debug logging for the integration with the following yaml configuration (and reboot after saving this config snippet)
```
logger:
  default: warn
  logs:
     custom_components.victron: debug
     homeassistant.components.modbus: debug
```
Please attach the logging generated after rebooting so I can better help you with this issue

closed 2023-03-06 by sfstar