sfstar/hass-victron #35: Victron reboot requires integration restart

cm2962, 2023-01-27:
Hi sfstar,
first, many thanks for your work and effort here. Highly appreciated.

I observe the situation that I need to restart your integration everytime I reboot my victron (raspi) device. Otherwise I don't get any values anymore.

Cheers
Christian

mtucker502, 2023-01-31:
I've noticed something similar. After enabling debug logging though it seems to work after reboot:

```
2023-01-31 16:23:54.355 DEBUG (MainThread) [custom_components.victron.coordinator] Finished fetching victron data in 3.081 seconds (success: False)
### Cerbo GX is rebooted....
2023-01-31 16:24:46.747 DEBUG (MainThread) [custom_components.victron.coordinator] Finished fetching victron data in 1.474 seconds (success: True)
```

Even after the reboot though total solar yield (which I use to track solar production in the energy dashboard is not available. I believe this is related to the fact that I now have duplicate entities:

[image]
[image]

It seems they're unavailable values are from restored entities:
[image]

What's strange is in the current state the solar yield will sometimes produce values until the Cerbo GX is rebooted. Then the value switches to the _100 entities which is when I notice the dashboard no longer working.

I removed the old entities, updated my dashboard to use the _100 entities. With debug still enabled, I rebooted the Cerbo GX:

```
2023-01-31 16:40:21.276 ERROR (MainThread) [custom_components.victron.coordinator] Error fetching victron data: Fetching registers failed
2023-01-31 16:40:21.277 DEBUG (MainThread) [custom_components.victron.coordinator] Finished fetching victron data in 0.003 seconds (success: False)
### Cerbo GX is rebooted....
2023-01-31 16:41:08.869 INFO (MainThread) [custom_components.victron.coordinator] Fetching victron data recovered
2023-01-31 16:41:08.869 DEBUG (MainThread) [custom_components.victron.coordinator] Finished fetching victron data in 1.596 seconds (success: True)
```

After rebooting, everything appears normal. I reloaded the integration as a test. I also disabled and re-enabled the integration.

I disabled debug logging and performed the same steps again.

```
2023-01-31 16:45:12.132 DEBUG (MainThread) [custom_components.victron.coordinator] {'100': ['battery_registers', 'battery_detail_registers', 'solarcharger_registers', 'solarcharger_tracker_voltage_registers', 'solarcharger_tracker_registers', 'settings_registers', 'settings_ess_registers', 'system_registers', 'system_battery_registers', 'system_dc_registers', 'system_charger_registers', 'system_power_registers', 'system_bus_registers'], '227': ['vebus_registers']}
2023-01-31 16:45:12.520 DEBUG (MainThread) [custom_components.victron.coordinator] Finished fetching victron data in 0.389 seconds (success: True)
### Cerbo GX is rebooted....

```

And everything still works... As I go back in my energy dashboard's history now using the `sensor.victron_solarcharger_yield_today_100` entity exclusively I can see days where the _100 entity was used and some days where the _0 entity was used by the integration.

For now I'll keep monitoring but hopefuly deleting the duplicates (_0 suffix entities in my case) will work unless the integration creates thema gain.

cm2962, 2023-03-02:
Seems to be solved for me now; just re-installed my Victron Raspbi

closed 2023-03-02 by cm2962