sfstar/hass-victron #36: Multiple PV Inverter in HomeAssistant

benedictjaeger, 2023-01-29:
Hi,
I run 3 PV inverters on my Victron Multiplus. These are also recognized in Venus and have different VRM instances (34,35,36).
However, when I run a scan in HomeAssistant, only the first inverter (34) is detected. I can recognize this from the displayed values.
Any solution for this?
BR
Ben

sfstar, 2023-02-01:
Hello and thank you for opening this issue.
I haven't tested the integration with multiple PV inverters present.
The specification for the modbus service states the following unit id's to be valid id's for modbus queryable devices
[image]

As can be seen in the provided screenshot the id's of these PV inverters should have been in the range of 20 until 29.
Unit id 34 is a valid id for a gavazzi meter but not a pv inverter instance.
However, the integration currently doesn't enforce only searching for meters on unit id 34.
That's why only 1 of the PV inverters gets detected.
On the one hand adding the undefined unit ids 35 and 36 to this integration should solve your issue.
But on the other hand there are increasingly more and more cases reported in which real world situations don't match with the specification.
Rather than patch hunting each issue i'm trying to gather a list of specification divergences to present to victron.
So that the modbus document is the definitive specification for this integration.
Since, patching and diverging from the specification won't ensure that the integration will keep working once victron releases another patch

benedictjaeger, 2023-02-01:
Thanks for this solution. 3 Inverters know are found in HA.  Would be great to provide this table also in the documentation. Now its dark outside and I will test the connection tomorrow when the inverters start producing.

sfstar, 2024-03-29:
The release (v0.2.0) has been shipped.
Closing issue, feel free to re-open if the issue persists with the new release

closed 2024-03-29 by sfstar