sfstar/hass-victron #37: vebus mode control & status problem?

dpa68, 2023-01-30:
Thank you sfstar. Great work. Took a few minutes to install and works great the last 3 days. Only problem I found is that: the vebus mode, when set to OFF, it does not work and the vebus mode (charger - inverter - on - off) does not refresh (it does not show the correct mode, when changed from HA). 
Latest HA version. Only one Multiplus (f/w 497), a Daly integrated bat (f/w 0.14.3) and an RPI4 with the v.2.92 venusOS, configured as UPS and load diverter here.

dpa68, 2023-02-01:
EDIT: vebus state returns the actual state of the Multiplus

sfstar, 2023-05-05:
Thank you for opening this issue.
From what I gather of your last comment on this issue it seems that the functionality of the integration is working correctly but that it might take until the multiplus changes it's state for it to reflect in the entity.
Is this correct?

dpa68, 2023-05-07:
Thank you for your response,

 

-          vebus mode “control”: does not work, when you click from inverter to ON. It will not change.

(vebus mode control does works, if you use it at an automation)

 

 

 

-          vebus state returns the actual state of the Multiplus There is only some delay of 5sec in refreshing, due to communication delay, I suppose (set at 1 sec)– no problem

-           

Thank you again

closed 2025-03-05 by github-actions[bot]