sfstar/hass-victron #38: Unit of measure on Tank

Next9999, 2023-02-01:
PLease mofidy the Unit of measure to 3 digit or more

com.victronenergy.tank | Tank capacity | 3001 | uint32 | 10000 | 0 to 429496.7296
-- | -- | -- | -- | -- | --

</body>

</html>

[image]

sfstar, 2023-02-01:
Hello and thank you for opening this issue.
Both the remaining and the capacity registers have the same decoding scheme definition (as can be seen in the screenshot).
[image]
Since the remaining sensor does report values with 3 digits after the . it seems that the capacity is 0.100 and it is being shorted by ha to 0.1.
Could you confirm if the capacity is set to 0.1m3 in the gx device configuration?

In order for me to further troubleshoot this I will need to setup a test venus OS device.
However, currently I do not have a lot of time available so this will have to wait until the end of februari / beginning of march

sfstar, 2024-03-15:
I believe this might be fixed by changing the display precision setting in home assistant.
Since it defaults to displaying 1 number after the decimal.
Changing this to a precision of 3 numbers after the decimal should provide you with what you are looking for.

closed 2025-03-05 by github-actions[bot]