sfstar/hass-victron #39: HA version 2022.2.0 with hass-victron v0.0.7

flyrmyr, 2023-02-01:
Just updated my HA core instance to the one released today. I didn't receive a HACS update option for this Victron integration, and on `redownload` I'm seeing this.

[image]

Attempting to continue anyway just hangs on the updating spinner.

sfstar, 2023-02-02:
Thank you for reporting. 
This was a typo in the hacs.json file. 
It should now be resolved. 
However, hacs might take a reboot or some time for this to reflect in the available updates

sfstar, 2023-02-05:
I've currently installed v0.0.7 on two systems after my previous comment.
Therefore I'm closing this issue as resolved.
Feel free to re-open if you're still experiencing this issue on your end

closed 2023-02-05 by sfstar