sfstar/hass-victron #40: state class 'measurement' which is impossible considering device class ('volume')

Next9999, 2023-02-02:
Tks again for all the work
i removed the integration, reload the new HACS and reinstall the integration.

========================================

Logger: homeassistant.components.sensor
Source: components/sensor/__init__.py:503
Integration: Sensor (documentation, issues)
First occurred: 08:25:20 (4 occurrences)
Last logged: 08:25:21

Entity sensor.victron_tank_capacity_20 (<class 'custom_components.victron.sensor.VictronSensor'>) is using state class 'measurement' which is impossible considering device class ('volume') it is using; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
Entity sensor.victron_tank_remaining_20 (<class 'custom_components.victron.sensor.VictronSensor'>) is using state class 'measurement' which is impossible considering device class ('volume') it is using; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
Entity sensor.victron_tank_capacity_21 (<class 'custom_components.victron.sensor.VictronSensor'>) is using state class 'measurement' which is impossible considering device class ('volume') it is using; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
Entity sensor.victron_tank_remaining_21 (<class 'custom_components.victron.sensor.VictronSensor'>) is using state class 'measurement' which is impossible considering device class ('volume') it is using; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.

sfstar, 2023-05-05:
Related to #62

sfstar, 2023-06-10:
Closing issue as resolved with release v0.0.10.
Feel free to re-open is the issue persists in your setup

closed 2023-06-10 by sfstar