sfstar/hass-victron #41: ValueError: 0 is not a valid alternator_errorcode

Next9999, 2023-02-02:
Tks again for all the work
i removed the integration, reload the new HACS and reinstall the integration.
=============================================

Logger: homeassistant.components.sensor
Source: custom_components/victron/sensor.py:155
Integration: Sensor (documentation, issues)
First occurred: 08:25:21 (1 occurrences)
Last logged: 08:25:21

victron: Error on device update!
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 507, in _async_add_entity
    await entity.async_device_update(warning=False)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 739, in async_device_update
    await task
  File "/config/custom_components/victron/sensor.py", line 155, in async_update
    self._attr_native_value = self.entity_type.decodeEnum(data).name.split("_DUPLICATE")[0]
  File "/usr/local/lib/python3.10/enum.py", line 385, in __call__
    return cls.__new__(cls, value)
  File "/usr/local/lib/python3.10/enum.py", line 710, in __new__
    raise ve_exc
ValueError: 0 is not a valid alternator_errorcode

sfstar, 2023-05-05:
The victron modbus TCP spec sheet references only the following return codes for that register address:
`12=High battery temperature;13=High battery voltage;14=Low battery voltage;15=VBat exceeds $CPB;21=High alternator temperature;22=Alternator overspeed;24=Internal error;41=High field FET temperature;42=Sensor missing;43=Low VAlt;44=High Voltage offset;45=VAlt exceeds $CPB;51-52=Battery disconnect request;53=Battery instance out of range;54=Too many BMSes;55=AEBus fault;56=Too many Victron devices;58-61=Battery requested disconnection;91=BMS lost;92=Forced idle;201=DCDC converter fail;201-207=DCDC error`
Therefore, it seems that this is another instance of the documentation not matching real world modbus server implementation by victron.

sfstar, 2024-03-29:
Hello @Next9999,

Would it be possible for you to find out the alternator status in the GX device when the errorcode 0 is returned?
So that we can implement this undocumented state into the integration.

closed 2025-03-05 by github-actions[bot]