sfstar/hass-victron #43: device class voltage, state class None and unit thus indicating it has a numeric value; however, it has the non-numeric value: OK (<class 'str'>)

Next9999, 2023-02-02:
Tks again for all the work
i removed the integration, reload the new HACS and reinstall the integration.

=============================================
Logger: homeassistant.components.sensor
Source: components/sensor/__init__.py:605
Integration: Sensor (documentation, issues)
First occurred: 08:25:20 (77 occurrences)
Last logged: 08:25:21

Sensor sensor.victron_inverter_alarm_lowacoutvoltage_239 has device class voltage, state class None and unit thus indicating it has a numeric value; however, it has the non-numeric value: OK (<class 'str'>); Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
Sensor sensor.victron_inverter_alarm_overload_239 has device class voltage, state class None and unit thus indicating it has a numeric value; however, it has the non-numeric value: OK (<class 'str'>); Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
Sensor sensor.victron_inverter_alarm_ripple_239 has device class voltage, state class None and unit thus indicating it has a numeric value; however, it has the non-numeric value: OK (<class 'str'>); Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
Sensor sensor.victron_inverter_info_state_239 has device class voltage, state class None and unit thus indicating it has a numeric value; however, it has the non-numeric value: INVERTING (<class 'str'>); Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
Sensor sensor.victron_inverter_alarm_lowsoc_239 has device class voltage, state class None and unit thus indicating it has a numeric value; however, it has the non-numeric value: OK (<class 'str'>); Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.

sfstar, 2023-02-03:
Should be addressed by the same fix as for #42

sfstar, 2023-02-03:
v0.0.8 is now released with a fix for this issue.
Please install it to resolve this issue.
For now I will be closing this issue report in favor of #25 since it has the same root cause.
Feel free to re-open if the issue isn't resolved after updating and restarting.

closed 2023-02-03 by sfstar