sfstar/hass-victron #44: ValueError: 0 is not a valid ess_mode

Next9999, 2023-02-02:
Tks again for all the work
i removed the integration, reload the new HACS and reinstall the integration.

=============================================
Logger: homeassistant.components.select
Source: custom_components/victron/select.py:127
Integration: Select (documentation, issues)
First occurred: 08:25:21 (2 occurrences)
Last logged: 08:25:21

Error adding entities for domain select with platform victron
Error while setting up victron platform for select
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 442, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 717, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 814, in add_to_platform_finish
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 566, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 607, in _async_write_ha_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 572, in _stringify_state
    if (state := self.state) is None:
  File "/usr/src/homeassistant/homeassistant/components/select/__init__.py", line 97, in state
    if self.current_option is None or self.current_option not in self.options:
  File "/config/custom_components/victron/select.py", line 127, in current_option
    return  self.description.options(self.description.value_fn(self.coordinator.processed_data(), self.description.slave, self.description.key)).name
  File "/usr/local/lib/python3.10/enum.py", line 385, in __call__
    return cls.__new__(cls, value)
  File "/usr/local/lib/python3.10/enum.py", line 710, in __new__
    raise ve_exc
ValueError: 0 is not a valid ess_mode

huganega, 2023-02-22:
same error too -> VenusOS 2.93

sfstar, 2023-05-05:
This is another instance of the victron modbus documentation not matching the real world output of the modbus server.
Valid outputs (according to the latest released modbus spec) are:
`1=ESS with Phase Compensation;2=ESS without phase compensation;3=Disabled/External Control`

sfstar, 2024-03-29:
Hello @Next9999 and @huganega,

Would it be possible for you to report the state of ess (on the GX device) when ess mode 0 is returned?
So that we can add it as an state to the integration?

pos-ei-don, 2024-03-29:
Is this list, extracted from the current Victron-Nodered Node not
complete? At least, I have never seen a mismatch on the GX.

> Value types
> 
> 0 - Unused, BL disabled
> 1 - Restarting
> 2 - Self-consumption
> 3 - Self-consumption
> 4 - Self-consumption
> 5 - Discharge disabled
> 6 - Force charge
> 7 - Sustain
> 8 - Low Soc Recharge
> 9 - Keep batteries charged
> 10 - BL Disabled
> 11 - BL Disabled (Low SoC)
> 12 - BL Disabled (Low SOC recharge)
>

sfstar, 2024-03-29:
Hello @pos-ei-don, 

Unfortunately these values are the values for the settings_ess_batterylife_state register, not the settings_ess_mode register

sfstar, 2024-05-19:
The issue has been shipped in the latest release (v0.3.0) closing issue as resolved.
Feel free to re-open if the issue persists with the v0.3.0 release

closed 2024-05-19 by sfstar