sfstar/hass-victron #45: Missing 1-wire sensor

mapeje, 2023-02-02:
I am using RpiGpioSetup and have a DS18B20 OneWire temperature sensor connected. It shows up in VRM as  "VRM instance 101" but it does come through to Home Assisstant. I only see my Ruuvi temperature sensor, VRM instance 20.

sfstar, 2023-02-03:
Thank you for reporting this issue.
Does the DC18B20 also show up in the "available services" section for the modbus service on the GX device?

mapeje, 2023-02-04:
Yes, it shows Unit ID 101, see below. For the HA integration, I only get "1", "20", "100"
[image]
[image]
[image]
[image]

sfstar, 2024-04-24:
Hello @mapeje,

Sorry for the long wait between your report and your issue being picked up.
The change that should allow your 1-wire sensor to be detected by the integration is now pushed to main.
Please install the main version of the integration and check if the device is detected correctly after a rescan.

sfstar, 2024-05-19:
The issue has been shipped in the latest release (v0.3.0) closing issue as resolved.
Feel free to re-open if the issue persists with the v0.3.0 release

closed 2024-05-19 by sfstar