sfstar/hass-victron #46: Device that becomes unavailable does not update correctly afterwards

braamvh, 2023-02-06:
This was originally another issue, but now that I have some experience with the integration, I thought I'd open a clean item.

In my case, I have a ET112 grid meter. When the mains is cut off (South Africa and loadshedding - this happens multiple times a day), the device does not get reported by modbus. When the device returns, subsequent queries does display it again. I tested this with my own modbus configuration.

However, the integration does not update the entities afterwards. It does show a fixed value, not changing, and the statistics says "updated x hours ago".  Other devices continue to update as normal. If I reload the integration, it starts to update correctly.

Currently I am working around this by having an automation that does a reload as soon as mains power is restored. I will monitor for a while to see if this works correctly, but so far it does.

I have replaced my CCGX with a Cerbo GX since the last report, but still get the same behaviour, so it's not hardware in the controller, as I thought before.

braamvh, 2023-02-06:
As a side note, you get the same behaviour from any entity that doesn't change in value (e.g. vebus state, or solar yield at night). HA shows it as last updated when the value changed, and resumes updates when the value changes again. So I believe the real issue here is that the integration doesn't update the value any more after it becomes available again, for some reason.

angyp, 2023-02-09:
I agree I have a PV inverter that goes missing due to a poor wifi connection. When it comes back I get the same behaviour the integration doesn't update. It's interesting because the VRM portal starts updating again so Venus has the right data. The only way to resolve it is to reload the integration.

mmaritz, 2023-02-24:
> I agree I have a PV inverter that goes missing due to a poor wifi connection. When it comes back I get the same behaviour the integration doesn't update. It's interesting because the VRM portal [...]

Same here - I have to reload the integration a number of times a week.  For example:  MPPT Max Yield from previous day didn't reset from yesterday.  
[image]
or
[image]

MesserschmittX, 2023-02-24:
I have a similar issue, there are two solar chargers in my system, the integration shows them as model 0 and model 1. Model 0 seems to go missing as soon as there is no solar power and then it will stay unavailable in homeassistant until I delete the whole integration and set it up again. (A reload does not bring it back for me) To me it seems like the integration is discarding devices that are no longer present and does not bring them back if they get available again

VieiraTino, 2023-03-05:
Yea, i have the same behaviour here.
This happens in mine "pvinverter" device. And since that is the only way i have to get the full solar production i cant get that info to home assistant. I have the "vebus" device that has the outtoinverter entity but that is only the produced that goes to the battery. It is missing the produced and used in my house.

sfstar, 2023-06-10:
I believe I narrowed the cause of the behaviour down to the coordinator returning data when it shouldn't.
I'm hoping to release a version in the coming weeks that better handles wifi drops, GX device reboots etc.
please stay tuned

braamvh, 2023-07-03:
Thanks for the connection update @sfstar. Looking forward to testing this.

I have disabled my workaround automations and will be checking how it goes. Though it will take a while, loadshedding here has been reduced lately and I am going on holiday soon.

sfstar, 2023-07-05:
@braamvh Thank you for taking the time and effort for testing the latest connection changes.

braamvh, 2023-07-14:
@sfstar For my devices the issue has completly gone away - in the past 2 weeks I could not find a single case where data stopped being collected. Thanks for the efforts in getting this resolved.

closed 2023-07-14 by braamvh