sfstar/hass-victron #48: changing number.victron_settings_ess_maxdischargepower error: required argument is not an integer

ChristophPech, 2023-02-08:
When changing the number.victron_settings_ess_maxdischargepower slider in the UI, I get an error message that the required argument is not an integer because it's a floating point number.

cm2962, 2023-02-09:
I observe the same. 
[image]

andzie82, 2023-02-28:
Hi, is this related to my issue as well? #52

ChristophPech, 2023-03-01:
Not really. I added configurable steps size in the code, but you need configurable range in the UI. Most people don't want to have a range larger than necessary because then the UI becomes difficult to use, outright impossible on mobile even.

andzie82, 2023-03-01:
I don’t know if I did something wrong when setting it up but I first put in 20A but I changed it to 60A instead. I guess it doesn’t take in effect if you have a 3 phase system or not.

sfstar, 2023-03-06:
This issue should now be adressed in the latest release containing the fix made by @ChristophPech.
I will close this issue if there are no further reports in the coming weeks

closed 2023-05-05 by sfstar