sfstar/hass-victron #51: More readings from BMS, e.g. Cell voltages/deviation

dominikfe, 2023-02-24:
Great integration, thank you!!!
Can I raise an improvement request?
Would be great, if the integration would read out BMS values like cell voltages or cell voltage deviations because the whole DIY community tracks their cell voltages carefully. Currently I need to read them via mqtt but doing this with your integration would be a lot easier...

braamvh, 2023-02-25:
The integration can only provide the information provided by the GX device over Modbus. I had a look through the spec and while minimum/maximumn cell voltage is provided, it is set to 0 on a managed battery. The full list is here:

[Modbus registers list](https://www.victronenergy.com/support-and-downloads/technical-information)

This is the same as the MQTT data available, so it is not available via either side. If Victron can add it to the list, it can be exposed via the integration.

pos-ei-don, 2023-02-25:
Cell Voltage is available in Mqtt, it can
 also be displayed in the console for certain battery types.

[image]

braamvh, 2023-02-26:
Yes, they can, but my point was that this is a modbus integration, so only values available via the modbus registers are available. It might be possible for some, but not a general thing that can be easily made available.

sfstar, 2023-03-06:
Adding won't fix for now since this isn't currently exposed by victron as mentioned by @braamvh.
@pos-ei-don if you want to open a feature request for this on the victron forum to get this added feel free to link this issue.
If not this issue will be archived after 01-04-2023 if no further activity occurs

sfstar, 2023-05-16:
This is however dependant on victron's willingness to approve a pr I will be making

closed 2025-03-05 by github-actions[bot]