sfstar/hass-victron #52: Change max/min number.victron_settings_ess_acpowersetpoint

andzie82, 2023-02-28:
Is it possible to change the values to the full range?

This register has range -32768W to 32767W (scale factor=1). Use register 2703 for a larger range (scale factor=0.01).

andzie82, 2023-02-28:
As of now it’s -5100 - 5100

sfstar, 2023-03-06:
It is calculated (as a safety precaution) based on your systems max amperage.
If you want a larger range you can just set the config setting to something extremely large.
However this is not advised since you will effectively remove the safeguards that prevent too large current to be triggered by this integration.
Proceed at your own peril

andzie82, 2023-03-06:
Yes i saw that. However since I have a three phase system I put in 60A instead of 20A and i have per unit. I didn’t know if it should be per phase or total.

sfstar, 2023-05-06:
This will be improved upon by the same changes that are required to address #67.
By improving the configuration setup so that you can specify the phasing setup this should become more self-explanatory.

sfstar, 2023-05-09:
Please check out v0.1.0 to try out the new configuration flow with support for multi phase systems.
Feel free to comment on any issues (or if it works correctly) found in this release regarding 3 phase support

sfstar, 2023-06-10:
Closing this issue since the v0.1.0 release has been released over a month ago and since there haven't been any three phase issues reported.
Feel free to re-open the issue if you feel that it persists/wasn't fully addressed

closed 2023-06-10 by sfstar