sfstar/hass-victron #54: Changing values not possible (eg. ess_acpowersetpoint)

cm2962, 2023-03-02:
I see changes when I act in VRM. 

But changes in your integration make values always jumping back to zero. Adjustments are not possible:

[image]

pos-ei-don, 2023-03-02:
This is a security feature of victron. You have to reset the value every couple of seconds (60s i think).
So if you want to use this make shure to enable a trigger to repeat setting this value.

michaelkuty, 2023-03-04:
nope, I think its fixed in https://github.com/sfstar/hass-victron/pull/50

cm2962, 2023-03-04:
Thank you Michael. What do I need to to to get this up and running?

sfstar, 2023-03-06:
Install v0.0.9 that was just released.
If it resolves your issue feel free to close this issue

cm2962, 2023-03-06:
Hi sfstar, first of all many thanks. 

Unfortunately, it does not work. Same behavior as before. 

Any changes I make result in "0". 

[image]

angyp, 2024-05-28:
I had a similar issue. In the integration configuration did you set the current? If not you will need to so the min and maximum values can be setup.

closed 2025-03-05 by github-actions[bot]