sfstar/hass-victron #56: vebus disappeared - reappeared with different ID

tobox, 2023-03-06:
Some of my victron readings in home assistent were suddenly unavailable. After restarting home assistant and rescanning the victron addon, it found a second vebus instance (238 instead of 239). I have only one victron multiplus ii, connected to a raspberry pi with venus os. Any Ideas how this happened? I had to change all my automations and lovelace panels to use the new instance.

[image]

sfstar, 2023-03-06:
Hello and thank you for opening an issue.
Unfortunately this cannot be resolved at the integrations end since the venus OS modbus server changed the id.
This also happened to me with one of my 2 gavazzi meters.
I fixed this meter by manually changing the xml configuration of the gx device (via root access).

tobox, 2023-03-07:
I dug into this problem and found the root cause: after finishing up the whole installation I properly fixed all wires and replugged all devices into the raspberry pi. If you change the USB port of the MK3 interface, a new device will be created (and the old one will be orphaned). 

So make sure to always plug everything into the same USB ports. 

Maybe you moved you RS485 adapter for the gavazzi meters to a different port as well?

Again, this issue can be closed because it is a "works as designed" thing inside venus os. 

The device appeared twice on VRM as well!

closed 2023-03-13 by tobox