sfstar/hass-victron #57: Using numeric inputs instead of sliders

tobox, 2023-03-06:
I find the sliders difficult to use for most of the victron inputs. Sometimes the step size is too large, sometimes just swiping on the mobile phone moves important sliders to their min or max values. I would rather use regular, numeric input fields...

pos-ei-don, 2023-03-06:
+1, fully agree. I think, sliders are dangerous.

sfstar, 2023-03-06:
I would feel more for making this configureable through the settings of the integration.
Rather than changing it to numeric input for everybody.
I'm open to a PR for this change. 
Otherwise it might take a while since I currently do not have a lot of spare time

sfstar, 2023-05-09:
Please try the release v0.1.0 this includes support to using numeric number entities by unchecking the newly introduced specific configuration option for slider mode.
Feel free to report back if there are any issues or if it's working as intended

randyoo, 2023-05-11:
> Feel free to report back if there are any issues or if it's working as intended

I've only used this to change the ESS grid setpoint, but it seems to work great for me. Thank you! The slider is handy, but it isn't nearly granular enough--in my case, it's in increments of 300W. If my finger moves even a bit while lifting, I could end up moving 3-4000W off target!

I think the ideal interface would be something like my hot water heater, perhaps with a configuration option to specify the increment of 1, 10, 50, 100W. (the GX device moves in 10W increments, by the way)

[image]

Not to be nit-picky, but I did also notice that number.victron_settings_ess_acpowersetpoint2, number.victron_settings_ess_batterylife_minimumsoc, number.victron_settings_ess_maxfeedinpower (and perhaps others?) show a ".0" (tenths of watts) precision:

[image]

sfstar, 2024-03-29:
Closing this issue as completed.
Tracking the tenths of a watt @randyoo reported in #196

closed 2024-03-29 by sfstar