sfstar/hass-victron #64: regular "lost connection" after update to 0.9

pos-ei-don, 2023-03-16:
Some values, especially my Victron mppt 3 and 4 stop sending Data to HA after several hours running.
I did not see this using 0.8 of this plugin.
I just revertet to check if this is solved with downgrading.

To reproduce this error, i just have to wait several hours. I see it when an MPPT gets a straigt line 
in ots power-value. Then it stopped updating this, while other Values still gets updated.
The Victron Console and the VRM still gets updated values from the device!

pos-ei-don, 2023-03-20:
This is stable in v0.0.8. Didn't have any disconnects the ast 3 days.

pos-ei-don, 2023-03-20:
Lost connection again today.
What I don't understand is why the orange Device recived data again after ~1hour while the purple didn't.
The VRM shows that both devices keept sending Data and still produced power.

So i think that this addon does not detect connection-disconnects and does nor reconnect automatically?
 When I manually "reload" the this integration, the data gets in again!

[image]

seidler2547, 2023-03-29:
I solved it for now with this automation:
```
alias: Reload victron integration
description: ""
trigger:
  - platform: time_pattern
    minutes: /1
condition:
  - condition: template
    value_template: >-
      {{ (((as_timestamp(now()) -
      as_timestamp(states.sensor.victron_battery_current_4.last_updated))  / 60)
      |  int) > 2 }}
action:
  - service: homeassistant.reload_config_entry
    data: {}
    target:
      entity_id: sensor.victron_battery_current_4
mode: single
```
(Replace _4 with the ID of your battery)

randyoo, 2023-04-09:
@seidler2547 said:
> I solved it for now with this automation:
 
Thanks for sharing! That's a helpful stop-gap, until this bug is solved. (It's affecting me, too)

seidler2547, 2023-04-16:
I actually changed to using MQTT, I get updates in real-time now with no load on the GX and automatic reconnect.

This is my repo where you can see my feeble attempts to make it work: https://github.com/seidler2547/victron-home-assistant-mqtt
(It works okay but is far from complete.)

pos-ei-don, 2023-04-16:
Perfect. Will take a look tomorrow! Thank you

randyoo, 2023-04-16:
> (It works okay but is far from complete.)

Appreciate you sharing your efforts. More options is always nice! It seems ESS/hub4 isn’t implemented, which I’m dependent upon—Using HA to update ESS grid power set point to target my AC PV production that’s on the AC input side of my Multis.

Hopefully the bugs will be worked out of this modbus implementation before too much longer!

pos-ei-don, 2024-03-17:
Works after Update

closed 2024-03-17 by pos-ei-don