sfstar/hass-victron #67: AC current limit setting doesn't account for split phase installs

randyoo, 2023-04-15:
Running a split phase configured pair of Multis, I entered the grid limit, in amps. In my case, it's 50. Used the "US" setting for AC voltage. As a consequence, the ESS max was set up to be 6000 watts.

By entering 100 amps (although inaccurate), I was able to set the expected maximum of 12000 watts. It may be possible that selecting "EU" rather than "US" would have been more appropriate, since the system is 240V between the two legs.

sfstar, 2023-05-05:
The integration doesn't currently support "special" configurations like split phase natively.
Using EU setting is a work-around you could use for now.
On the long term I need to revisit the configuration anyway (I will get three phase 230v soon) so that split phase and three phase configurations can more easily be configured.
Adding improvement label for now

randyoo, 2023-05-11:
> Please try out the new v0.1.0 release. This should now support the split phase configuration natively. Feel free to comment if you experience any issues with the update (or that everything is working without issues ofc).

I gave it a try. After deleting the integration, updating in HACS, then re-adding the integration again, it seemed to completely crash Home Assistant. After restarting the host machine, I was able to successfully install the new release, and configured my split phase system. Aside from hiccup with the initial crash, it seems to working well--thanks!

The only improvement I can think of would be **defaulting to hiding or disabling the "L3" entities** for split phase, and probably both L2 & L3 for the single phase installs.

In the process of trying to identify the cause of the crash when I first added the v0.1.0 release, I saw some other warnings in the logs. I'll soon open a separate issue for those, if they haven't already been identified.

closed 2023-05-11 by randyoo