sfstar/hass-victron #68: Error from Victron EV Charger

gb095666, 2023-05-04:
This error originated from a custom integration.

Logger: homeassistant.helpers.entity
Source: custom_components/victron/sensor.py:155
Integration: victron (documentation, issues)
First occurred: 15:33:55 (1 occurrences)
Last logged: 15:33:55

Update for sensor.victron_evcharger_status_40 fails
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 554, in async_update_ha_state
    await self.async_device_update()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 784, in async_device_update
    await coro
  File "/config/custom_components/victron/sensor.py", line 155, in async_update
    self._attr_native_value = self.entity_type.decodeEnum(data).name.split("_DUPLICATE")[0]
  File "/usr/local/lib/python3.10/enum.py", line 385, in __call__
    return cls.__new__(cls, value)
  File "/usr/local/lib/python3.10/enum.py", line 710, in __new__
    raise ve_exc
ValueError: 21 is not a valid evcharger_status

sfstar, 2023-05-05:
Hello and thank you for opening this issue.
What is the status of the charger in the victron environment when status code 21 is returned?
The documentation doesn't reference a status code of 21

gb095666, 2023-05-05:
Hi,
There seems to be another modbus spreadsheet for the EV charger which has more information in, not sure why it's not been incorporated into the latest 2.90 version.

If you follow the link at the bottom of this post you will see version 2.90 around 15 lines down, just below that there is another specifically for the EV charger, "Modbus-TCP register list EVCS", within there unit address 5015 seems to be the Charger status, with 21 being defined as "Start Charging"

Hope this helps, Can I also point out that there is another discrepancy between both spreadsheets, with 5026 Charger position, and 3827 in version 2.90, if you look at "Inverter in" and "inverter out" the numbers have been transposed, I can confirm that the 5026 entry is correct, do you want me to open another bug fix for this ?

https://www.victronenergy.com/support-and-downloads/technical-information

gb095666, 2023-08-22:
Hi,

Can you let me know please when this will be incorporated into the current release.

sfstar, 2024-03-14:
Hello,

Sorry for the delay in looking at your PR.
Just to clarify, the values that are supposed to be returned for register 5026 (according to the evcs register list) are also being reported via the 3827 register?
Are they both kept in sync or is 5026 the replacement of 3827?

gb095666, 2024-03-15:
Hi,

They are both keep in sync.
The EVCS has an inbuilt Modbus server, so if you query the EVCS directly it will respond on 5026, if you query the GX device it will respond on 3827, with the same value.

The CCGX-Modbus-TCP-register-list-3.20.xlsx spreadsheet description is wrong for address 3827, I have raised a bug ticket with Victron.

closed 2025-03-05 by github-actions[bot]