sfstar/hass-victron #69: non numeric value sensors became unavailable

dbachats, 2023-05-04:
Hi since this morning all the non numeric value sensors became unavailable across the integration. thanks for the good work thanks for the help
[image]

Linusten, 2023-05-04:
I am facing also These issues since i updated to homeassistant 2023.5.0

sfstar, 2023-05-05:
Thanks for opening this issue.
It seems to have been introduced in homeassistant 2023.5.0.
I've also seen reports of built-in integrations that now have failing entities with the 2023.5 release.
Will investigate if this can be resolved on the integrations end
```
Logger: homeassistant.components.websocket_api.http.connection
Source: components/sensor/__init__.py:583
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 8:19:38 AM (1 occurrences)
Last logged: 8:19:38 AM

[140318193989952] Sensor sensor.victron_pvinverter_position_30 has device class None, state class None unit and suggested precision None thus indicating it has a numeric value; however, it has the non-numeric value: AC_OUTPUT (<class 'str'>)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 579, in state
    numerical_value = int(value)
ValueError: invalid literal for int() with base 10: 'AC_OUTPUT'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 199, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1849, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1889, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 226, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 798, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 981, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 838, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 106, in async_set_value
    await entity.async_set_native_value(native_value)
  File "/config/custom_components/victron/number.py", line 191, in async_set_native_value
    await self.coordinator.async_update_local_entry(self.data_key, int(value))
  File "/config/custom_components/victron/coordinator.py", line 131, in async_update_local_entry
    self.async_set_updated_data(data)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 376, in async_set_updated_data
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 146, in async_update_listeners
    update_callback()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 406, in _handle_coordinator_update
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 585, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 649, in _async_write_ha_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 591, in _stringify_state
    if (state := self.state) is None:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 583, in state
    raise ValueError(
ValueError: Sensor sensor.victron_pvinverter_position_30 has device class None, state class None unit  and suggested precision None thus indicating it has a numeric value; however, it has the non-numeric value: AC_OUTPUT (<class 'str'>)

```

sfstar, 2023-05-05:
So the issue seems to have been created by a state_class of measurement being set on non numeric values like the pvinverter_position register.
See https://github.com/home-assistant/core/pull/87004 for more information.

A PR that was created by @nijel and was merged a couple of days ago resolves this issue for multiple entities but not all entities that are supposed to be set to a state_class of None were set to None in the PR #58 that resolves #22.
It seems that in 2023.5 home assistant didn't introduce a bug but rather no longer soft failed for an incorrectly registered entity.
I expect that I will be able to release a fix later today.

dbachats, 2023-05-05:
Super.  Thanks for taking the time for resolving this. Some entities are critical to solar installations controlled by automations. Thanks again for your work.

Linusten, 2023-05-05:
Thanks for the update, works perfectly :)

sfstar, 2023-05-05:
Thanks for the feedback.
Will leave this issue open the next couple of days in order to allow users to find this thread more easily.
After which I will close the issue as resolved (if no new instances of this bug are reported)

dbachats, 2023-05-05:
Just updated the integration and everything works perfectly again. I'll keep an eye on it. Thank you again.

sfstar, 2023-06-10:
Closing issue in accordance with my comment https://github.com/sfstar/hass-victron/issues/69#issuecomment-1536171708

closed 2023-06-10 by sfstar