sfstar/hass-victron #71: Multiple Logbook entries every 30 seconds

gb095666, 2023-05-05:
I have set the update interval for the Victron integration to 30 seconds

Within the Logbook option in HA and am seeing the following 2 entries repeating every 30 seconds.

I understand the values are changing, but don't understand why these are being echoed on the Logbook.

battery system maxvoltagecellid changed to 1
20:31:05 - Now
battery system minvoltagecellid changed to 3
20:31:05 - Now
battery system maxvoltagecellid changed to 3
20:30:35 - 30 seconds ago
battery system minvoltagecellid changed to 1
20:30:35 - 30 seconds ago
battery system maxvoltagecellid changed to 1
20:30:05 - 1 minute ago
battery system minvoltagecellid changed to 2
20:30:05 - 1 minute ago
battery system minvoltagecellid changed to 3
20:29:35 - 2 minutes ago
battery system maxvoltagecellid changed to 2
20:29:05 - 2 minutes ago

gb095666, 2023-05-05:
Also seeing:-

battery system mintemperaturecellid changed to 1
21:08:13 - 14 minutes ago

sfstar, 2023-05-06:
Hello and thank you for opening this issue.
This is not something the integration can control.
The logbook functionality of home assistant let's you see what entities changed when and what (automation, integration, user).
This allows users to troubleshoot wierd behaviour with their home automation.
Rapid changes in the min / max cell voltage id could happen when the battery is balancing (and in case of lifepo4 cells) being in the bottom / upper part of the charge curve.
Since not every cell charges at exactly the same rate.

This is not something I can fix but is rather the design decision of home assistant and the nature of the data being served by the victron modbus server.
Therefore, closing this issue.

closed 2023-05-06 by sfstar

gb095666, 2023-05-06:
Hi,

OK, I understand, I was just surprised that this is the first time its happened since upgrading HA to the latest version, in previous versions I never saw all these entries.

Thanks for replying.