sfstar/hass-victron #76: various numeric settings became unavailable since update to HA 2023.5.3

1984ozone1984, 2023-05-17:
Since update to HA version 2023.5.3 nearly all numeric settings became unavailable. Sensors are all working fine.

Thanks a lot for your help!
Stefan

[image]
<br>

[image]

sfstar, 2023-05-17:
Hello and thank you for opening this issue.
Have you also updated the integration recently.
If so please remove the integration (as instructed in the release notes) and configure the integration again.
The entities should come back with the same id's and no automations etc should be affected by this action.

If the issue persists after removing and configuring the integration then please let me know

1984ozone1984, 2023-05-17:
Hi,
thanks for your answer. Indeed the removal and reconfiguration of the integration solved the issue.
Sorry for the inconvenience of not fully reading the release notes!

closed 2023-05-17 by 1984ozone1984