sfstar/hass-victron #77: Since new version 0.1.0 variables in HA do not update

mroggi, 2023-05-21:
I activated the debug function and get the following error:

```
2023-05-21 13:26:58.783 DEBUG (MainThread) [custom_components.victron.coordinator] Manually updated victron data
2023-05-21 13:27:01.471 ERROR (MainThread) [async_upnp_client.ssdp] Received error: [Errno 19] No such device, transport: <_SelectorDatagramTransport fd=24 read=polling write=<idle, bufsize=0>>, socket: <asyncio.TransportSocket fd=24, family=AddressFamily.AF_INET, type=SocketKind.SOCK_DGRAM, proto=0, laddr=('0.0.0.0', 44135)>
2023-05-21 13:27:01.472 ERROR (MainThread) [async_upnp_client.ssdp] Received error: [Errno 101] Network unreachable, transport: <_SelectorDatagramTransport fd=29 read=polling write=<idle, bufsize=0>>, socket: <asyncio.TransportSocket fd=29, family=AddressFamily.AF_INET6, type=SocketKind.SOCK_DGRAM, proto=0, laddr=('::', 45698, 0, 0)>
```

It seems to sporadically work, like every couple of hours, but of course does not help.
Any idea what I need to do to get updated values?
Thanks,
mrogi

closed 2023-05-21 by mroggi

mroggi, 2023-05-21:
Reconfigured and started working again

sfstar, 2023-05-26:
This seems like a local network issue (that the data cannot be send due to the local network being unavailable).
Great to hear that reconfiguring fixed this but the integration can't "fix" this issue.
Since it is located outside of the integrations software.